mattanger/ckb-next #304: Building latest version with ./quickinstall fails with errors in output. error: use of undeclared identifier 'exit_early'

jaakkovarjo, 2017-09-30:
# Bug report

- [x] Yes
- [ ] No

## General information
Building latest version with ./quickinstall fails. Quickinstall output shows errors relating to input_mac_mouse.c file. 

Below are the errors and included the full output if the quickinstall script in quickinstall_output.txt file. The errors start at line number 363 in the quickinstall_output.txt file.

[quickinstall_output.txt](https://github.com/mattanger/ckb-next/files/1346369/quickinstall_output.txt)

input_mac_mouse.c:482:25: error: use of undeclared identifier 'exit_early'
    require(parameters, exit_early);
                        ^
2 warnings generated.
input_mac_mouse.c:483:65: error: use of undeclared identifier 'exit_early'
    require(CFGetTypeID(parameters) == CFDictionaryGetTypeID(), exit_early);
                                                                ^
input_mac_mouse.c:521:31: error: use of undeclared identifier 'exit_early'
    require(parametricCurves, exit_early);
                              ^
input_mac_mouse.c:522:38: error: use of undeclared identifier 'exit_early'
    require(f_gt_sc(crsrScale, 0LL), exit_early);
                                     ^
input_mac_mouse.c:523:37: error: use of undeclared identifier 'exit_early'
    require(f_gt_sc(devScale, 0LL), exit_early);
                                    ^
input_mac_mouse.c:524:36: error: use of undeclared identifier 'exit_early'
    require(f_gt_sc(desired, 0LL), exit_early);
                                   ^
input_mac_mouse.c:533:17: error: use of undeclared identifier 'exit_early'
                exit_early);
                ^
input_mac_mouse.c:551:26: error: use of undeclared identifier 'exit_early'
        require(success, exit_early);
                         ^
input_mac_mouse.c:583:35: warning: 'bzero' call operates on objects of type 'IOHIPointing__PASecondaryParameters' while the size is based on a different type 'IOHIPointing__PASecondaryParameters *' [-Wsizeof-pointer-memaccess]
    bzero(secondaryParams, sizeof(secondaryParams));
          ~~~~~~~~~~~~~~~         ^~~~~~~~~~~~~~~
input_mac_mouse.c:583:35: note: did you mean to dereference the argument to 'sizeof' (and multiply it by the number of elements)?
    bzero(secondaryParams, sizeof(secondaryParams));
                                  ^~~~~~~~~~~~~~~
input_mac_mouse.c:1098:70: warning: comparison of integers of different signs: 'IOFixed' (aka 'int') and 'unsigned long long' [-Wsign-compare]
                _scrollWheelInfo.axis[type].isHighResScroll    = res > (SCROLL_DEFAULT_RESOLUTION * 2);
                                                                 ~~~ ^  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
input_mac_mouse.c:1217:55: warning: comparison of integers of different signs: 'IOFixed' (aka 'int') and 'unsigned long long' [-Wsign-compare]
                (scaleInfo->state.deltaTime[avgIndex] >= SCROLL_EVENT_THRESHOLD_MS)) {
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~~~~~~~~~~~~~~~~~~~
input_mac_mouse.c:1225:28: warning: comparison of integers of different signs: 'IOFixed' (aka 'int') and 'unsigned long long' [-Wsign-compare]
        if (avgTimeDeltaMS >= (SCROLL_CLEAR_THRESHOLD_MS_LL * kIOFixedOne)) {
            ~~~~~~~~~~~~~~ ^   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
input_mac_mouse.c:1237:24: warning: comparison of integers of different signs: 'IOFixed' (aka 'int') and 'unsigned long long' [-Wsign-compare]
    if (avgTimeDeltaMS > SCROLL_EVENT_THRESHOLD_MS) {
        ~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~
input_mac_mouse.c:1240:29: warning: comparison of integers of different signs: 'IOFixed' (aka 'int') and 'unsigned long long' [-Wsign-compare]
    else if (avgTimeDeltaMS < kIOFixedOne) {
             ~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~
input_mac_mouse.c:1403:35: warning: unused parameter 'event' [-Wunused-parameter]
static void do_setup(io_connect_t event, struct timespec now){
                                  ^
8 warnings and 8 errors generated.
make[2]: *** [debug/input_mac_mouse.o] Error 1

### Source
Cloned repository https://github.com/mattanger/ckb-next.git and built application manually using quickinstall script

### Branch
Master. Latest commit cfd3688

light2yellow, 2017-09-30:
What's the output of `sw_vers -productVersion`?

jaakkovarjo, 2017-09-30:
10.13

light2yellow, 2017-09-30:
I have just tried to update my Hackintosh from 10.12.6 and it died. So, not much I can help with.

PR #305 (2017-10-02): Fix daemon compilation issues in OSX 10.13.
The requires macro without leading underscores is no longer available by default. Also, the compiler now surfaces a warning which revealed a bug in the ported acceleration code.

Fixes #304.
closed 2017-10-02 by tatokis