mattanger/ckb-next #307: Corsair Scimitar Pro Issue

Sankhai, 2017-10-03:
Hi all,
Everything seem to be working fine with the app but the left click seems broken.
Sometimes I just can't click something, like there's no left click at all but then it works on something else. (F.E. I click on a TAB in google chrome, it works, then I try an another TAB : it won't ever load.)

It may sound stupid but it's completely impossible to use the mouse like that.
As anyone experienced this once?

Thank you in advance for your tips. 

_**Sankhai**
MBP 15" Late 2014
Graphic Designer_

PS: The left click is correctly assigned. 
[image]

light2yellow, 2017-10-03:
Can you reproduce this behaviour without `ckb-daemon` running?
 
You can disable it with `sudo launchctl unload -w /Library/LaunchDaemons/com.ckb.daemon.plist`.

Sankhai, 2017-10-04:
I just tried it and unfortunately now ckb just won't detect the mouse no matter what I try (plug/unplug/reboot) :( 
Thank you for trying to help!

[image]

MySpl33n, 2017-10-12:
Have you tried assigning left click to another button? Just to make sure it's not a software issue.

To rule out a hardware problem, does left click work on other computers/OSs? Or try assigning a different function to M1 and try that.

Sankhai, 2017-10-12:
Woaw I've finally figured out what was the problem.
#258 Topic helped me a lot. 
Seems like it's actually the MacOS Sierra "double-click speed" that stucks the left click.
Resolved it by putting the sensor a bit higher than mid range.
Closing the topic, thank you all.

closed 2017-10-12 by Sankhai

light2yellow, 2017-10-12:
@Sankhai thanks for sharing the solution, this is something to consider when dealing with mice issues on macOS.