mattanger/ckb-next #313: support for K55

y-raude-stjohns, 2017-10-12:
# New device support request

- [X] Yes
- [ ] No

-------
Is it possible / hard to add support for the K55 keyboard ?

Thx !

Here is the ckb-dev-detect-report :

OS Version:
ProductName:	Mac OS X
ProductVersion:	10.11.2
BuildVersion:	15C50

USB Devices:
                Corsair Gaming K55 RGB Keyboard:

                  Product ID: 0x1b3d
                  Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
                  Version: 10.18
                  Serial Number: AEFEA00358722F70F500194240048000
                  Speed: Up to 12 Mb/sec
                  Manufacturer: Corsair
                  Location ID: 0x14142000 / 15
                  Current Available (mA): 1000
                  Current Required (mA): 500
                  Extra Operating Current (mA): 0

light2yellow, 2017-10-13:
See #277.

closed 2017-10-13 by light2yellow

but2002, 2017-11-02:
Just noting, that I am an owner of this keyboard, and you -CAN- control the keyboard through the CUE software.

tatokis, 2017-11-23:
@but2002 what kind of control does CUE allow? Does it let you set the LEDs individually per key like the other devices?

but2002, 2017-11-24:
It is only by zone.  There are three zones it lets you control the color
of, but it does let you control them.

tatokis, 2017-12-01:
Tracking this in #277. If interested, subscribe to that.