mattanger/ckb-next #315: can't record a keybind in Discord on Mac

rkitover, 2017-10-14:
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1384605/ckb-dev-detect-report.gz)

[usb_bus_probe.txt](https://github.com/mattanger/ckb-next/files/1384608/usb_bus_probe.txt)

## What is the program's current behavior?

Cannot make a keybind in Discord on Mac, when I press "record keybind" and then press a key, nothing happens.

## What would you like it to do instead?

Be able to use Push-To-Talk with the keyboard in Discord.

## Any other notes & comments?

I can record a keybind with my other keyboard (that is also plugged in at the same time) but I haven't tried using it yet because I'm new to Discord and haven't found any voice chats yet.

# Bug report

- [x] Yes

## General information

### Source

built manually using `./quickinstall`

### Branch

master

eklundchristopher, 2017-10-18:
I'm in the exact same position. Using a SteelSeries M800 it works fine, but doesn't respond to anything that is pressed through my Corsair Strafe RGB.

I would also like to mention that the same goes for TeamSpeak. It is not just limited to Discord. However, registering a hotkey in native macOS functionality works (e.g. Dictation) or apps such as Alfred, Magnet and BetterSnapTool. I don't know if that helps or not.

Sush132, 2017-11-07:
Same problem here, when I use my wireless logitech keyboard it records the key bind but with my Corsair Strafe it doesn't seem to work.

SiirRandall, 2017-11-08:
For Mac just wait for the CUE Mac version to come out. I got the beta and it works great. It's a shame that this project got picked up by a bunch of Linux people. They even said they don't really care about the Mac version. This project was made for mac first then Linux. Now the new releases break more mac compatibility and just gets worse.  The new devs don't even care about mac support. Good news is CUE for macOS should be released soon and don't have to bother with this buggy software.

tatokis, 2017-11-08:
@SiirRandall 
First of all, you are being off topic.

Secondly, if we didn't care about mac, we wouldn't be having this thread open, nor constantly consider that every change we make needs to still work on mac and not break anything.

Additionally, you may have the money for an apple device, I do not. And I am not willing to spend that amount of money for hardware that is not going to improve my life, just to write code for free. And that's excluding everything else that is involved with dealing when developing for apple products. Sub-par function documentation being one of them.

Finally, this buggy software you are "dealing with" was made for free, out of people's spare time, out of passion and love. We have absolutely no connection with corsair, and they never helped the previous dev, nor us. If you do not appreciate it, it's fine, you are entitled to your opinion, but please do not try and insult us on our own issue tracker. Why didn't you bother trying to fix these bugs yourself?

Oh, and as a sidenote, this software wasn't "mac first". Check your sources before you spread false information.

SiirRandall, 2017-11-08:
It's not off topic. He is having issues with your software that you don't
even really support or the need the hardware to support or debug it. I told
him a solution that is coming. Maybe you should go back the corsair forums.
The original guy did his best to support both. You, on the other hand, are
putting Linux Support over Mac support and the original project wasn't all
about that. It may be free and it works great on Linux. If you can't
support mac people then just drop the support for it. You even said you
don't have the hardware to test it out. This is why there are so many bugs
with the macOS version. Not off topic, I am giving him a solution that will
work when Corsair releases CUE for macOS.

light2yellow, 2017-11-09:
@SiirRandall 

Although you are wrong on most of the aspects you've outlined, this is not the place to go through it.

Please, stop spamming (and trolling) here, otherwise your comments might get removed as they are clearly off-topic (use Google to lookup what this word means if you don't know). The unwritten rule is "KYS if you don't have anything to contribute [on the topic]". The topic of this ticket is: "can't record a keybind in Discord on Mac".

If you want to discuss anything with the community, proceed with the project's Freenode channel or the project's mailing list.

light2yellow, 2017-11-09:
@SirRandall

To escape confusion and misunderstanding, I'd like to add that KYS above means "keep yourself silent" and not anything else. In a private message I was told KYS means "kill yourself", but I couldn't even imagine it means that (abbreviation doesn't make the sense either, but whatever), I've always used it as a shortening for "keep yourself silent". So, my apologies. All I wanted to say is to don't post any noise if there's nothing to contribute on the topic, and start another topic when there's any new meta.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/38

closed 2018-01-22 by tatokis