mattanger/ckb-next #317: CKB-next don't start

SrSands, 2017-10-21:
@@
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1404199/ckb-dev-detect-report.gz)

Im on the latest version of kali and when i try to start ckb-next from the terminal nothing happens and i get 
`QObject::startTimer: Timers cannot have negative intervals
The ckb GUI should not be run as root.`
I follow the instructions as they appear

Can anyone help me? Thx

berolinux, 2017-10-31:
The "QObject::startTimer: Timers cannot have negative intervals" part is harmless.

The "The ckb GUI should not be run as root." part is what you need to pay attention to. Run it as regular user, and without sudo and friends.

tatokis, 2017-11-27:
Kali is not officially supported.
If you want to use it, you'll have to either not run as root, or remove the root check yourself.

closed 2017-11-27 by tatokis