mattanger/ckb-next #319: Color selector button causes program to crash

robfrawley, 2017-10-27:
- [Devices information](#devices-information)
- [New device support request](#new-device-support-request)
- [Feature request](#feature-request)
- [Bug report](#bug-report)
    - [General information](#general-information)
        - [Source](#source)
        - [Branch](#branch)
    - [Logs & crash reports](#logs--crash-reports)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior-1)
    - [What is the expected behavior of the program?](#what-is-the-expected-behavior-of-the-program)

---

# Devices information

[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1420969/ckb-dev-detect-report.gz)

---

# New device support request

- [ ] Yes
- [x] No
---

# Feature request

- [ ] Yes
- [x] No

---

# Bug report

- [x] Yes
- [ ] No

## General information

### Source

Built manually using `quickinstall` script on Ubuntu Linux 17.04.

### Branch

The `master` branch.

## Logs & crash reports

```
./bin/ckb
QObject::startTimer: Timers cannot have negative intervals
Path  to settings is "/home/rmf/.config/ckb/ckb.conf"
Scanning  "/usr/lib/ckb-animations/ckb-gradient"
Scanning  "/usr/lib/ckb-animations/ckb-heat"
Scanning  "/usr/lib/ckb-animations/ckb-pinwheel"
Scanning  "/usr/lib/ckb-animations/ckb-rain"
Scanning  "/usr/lib/ckb-animations/ckb-random"
Scanning  "/usr/lib/ckb-animations/ckb-ripple"
Scanning  "/usr/lib/ckb-animations/ckb-wave"
Starting  "/usr/lib/ckb-animations/ckb-gradient"
Downloaded new firmware list. 12 entries found.

(ckb:24094): GLib-GObject-WARNING **: cannot register existing type 'GtkWidget'

(ckb:24094): GLib-GObject-WARNING **: cannot add class private field to invalid type '<invalid>'

(ckb:24094): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(ckb:24094): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:24094): GLib-GObject-WARNING **: cannot register existing type 'GtkBuildable'

(ckb:24094): GLib-GObject-CRITICAL **: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed

(ckb:24094): GLib-CRITICAL **: g_once_init_leave: assertion 'result != 0' failed

(ckb:24094): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:24094): GLib-GObject-CRITICAL **: g_type_register_static: assertion 'parent_type > 0' failed

(ckb:24094): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'
```

## What is the program's current behavior?

Clicking the color selector button (after highlighting a number of keys) causes the program to stall and then pop up with a "Force Quit" or "Wait" dialoge. Waiting does not resolve the issue.

## What is the expected behavior of the program?

Clicking the color selector button should open the color selector pallet window.

emiliardo78, 2017-11-22:
Same problem too. After upgrade to ubuntu 17.10 che color selector freeze the whole application and waiting doesn't works. 
Thinked it was a problem with Gnome GDM but I have the same problem also with Unity (chenged GDM on the login screen)

dominic16y, 2017-11-24:
Same problem too. here is my system environment:
debian9 64bit, mate1.16.2

tatokis, 2017-11-24:
Upstream Qt bug, and duplicate.
https://github.com/mattanger/ckb-next#fedora-26-color-changer-freeze-fix
https://github.com/mattanger/ckb-next/issues/175#issuecomment-299453783

closed 2017-11-24 by tatokis