mattanger/ckb-next #321: Can't build on solus

QuillzLouis, 2017-10-28:
Trying to build on solus, failing. I have installed the system.devel package and qt5-base-devel, output from sudo ./quickinstall below.
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1424149/ckb-dev-detect-report.gz)

`quillztoxic@unfriended-blocked-pc ~/Downloads/ckb-next-master $ sudo ./quickinstall 
Password: 
Applying required qmake workaround
Preparing build files...
Compiling binaries...
(This can take a while, please be patient)
cd src/ckb-daemon/ && ( test -e Makefile || /usr/bin/qmake -o Makefile /home/quillztoxic/Downloads/ckb-next-master/src/ckb-daemon/ckb-daemon.pro ) && make -f Makefile 
cd src/ckb/ && ( test -e Makefile || /usr/bin/qmake -o Makefile /home/quillztoxic/Downloads/ckb-next-master/src/ckb/ckb.pro ) && make -f Makefile 
cd src/ckb-wave/ && ( test -e Makefile || /usr/bin/qmake -o Makefile /home/quillztoxic/Downloads/ckb-next-master/src/ckb-wave/ckb-wave.pro ) && make -f Makefile 
cd src/ckb-ripple/ && ( test -e Makefile || /usr/bin/qmake -o Makefile /home/quillztoxic/Downloads/ckb-next-master/src/ckb-ripple/ckb-ripple.pro ) && make -f Makefile 
make[1]: Entering directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb-daemon'
make -f Makefile.Debug
make[1]: Entering directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb-ripple'
make -f Makefile.Debug
make[2]: Entering directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb-daemon'
gcc -c -pipe -std=gnu11 -Wall -Wextra -fsigned-char -g -fPIC -DCKB_VERSION_STR=\"`cat /home/quillztoxic/Downloads/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o debug/device.o device.c
make[2]: Entering directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb-ripple'
gcc -c -pipe -std=c99 -g -fPIC  -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o debug/main.o main.c
make[1]: Entering directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb'
make -f Makefile.Release
make[2]: Entering directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb'
/usr/bin/uic mainwindow.ui -o ui_mainwindow.h
make[1]: Entering directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb-wave'
make -f Makefile.Debug
make[2]: Entering directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb-wave'
gcc -c -pipe -std=c99 -g -fPIC  -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o debug/main.o main.c
/usr/bin/uic kbwidget.ui -o ui_kbwidget.h
/usr/bin/uic settingswidget.ui -o ui_settingswidget.h
/usr/bin/uic kblightwidget.ui -o ui_kblightwidget.h
make[2]: Leaving directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb-daemon'
make[1]: Leaving directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb-daemon'
/usr/bin/uic kbanimwidget.ui -o ui_kbanimwidget.h
/usr/bin/uic animadddialog.ui -o ui_animadddialog.h
/usr/bin/uic animsettingdialog.ui -o ui_animsettingdialog.h
/usr/bin/uic gradientdialog.ui -o ui_gradientdialog.h
/usr/bin/uic kbprofiledialog.ui -o ui_kbprofiledialog.h
/usr/bin/uic kbbindwidget.ui -o ui_kbbindwidget.h
/usr/bin/uic rebindwidget.ui -o ui_rebindwidget.h
/usr/bin/uic modeselectdialog.ui -o ui_modeselectdialog.h
/usr/bin/uic fwupgradedialog.ui -o ui_fwupgradedialog.h
/usr/bin/uic mperfwidget.ui -o ui_mperfwidget.h
/usr/bin/uic kperfwidget.ui -o ui_kperfwidget.h
/usr/bin/uic layoutdialog.ui -o ui_layoutdialog.h
/usr/bin/uic extrasettingswidget.ui -o ui_extrasettingswidget.h
g++ -c -pipe -std=gnu++11 -Wall -Wextra -Wno-unused-parameter -fsigned-char -O2 -Wall -W -D_REENTRANT -fPIC -DCKB_VERSION_STR=\"`cat /home/quillztoxic/Downloads/ckb-next-master/src/ckb/../../VERSION | tr -d \n`\" -DQUAZIP_STATIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/QtWidgets -isystem /usr/include/QtGui -isystem /usr/include/QtNetwork -isystem /usr/include/QtCore -Irelease -isystem /usr/include/libdrm -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o release/main.o main.cpp
g++ -c -pipe -std=gnu++11 -Wall -Wextra -Wno-unused-parameter -fsigned-char -O2 -Wall -W -D_REENTRANT -fPIC -DCKB_VERSION_STR=\"`cat /home/quillztoxic/Downloads/ckb-next-master/src/ckb/../../VERSION | tr -d \n`\" -DQUAZIP_STATIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/QtWidgets -isystem /usr/include/QtGui -isystem /usr/include/QtNetwork -isystem /usr/include/QtCore -Irelease -isystem /usr/include/libdrm -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o release/mainwindow.o mainwindow.cpp
g++  -o ../../bin/ckb-animations/ckb-wave debug/main.o    
g++  -o ../../bin/ckb-animations/ckb-ripple debug/main.o    
make[2]: Leaving directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb-ripple'
make[1]: Leaving directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb-ripple'
g++ -c -pipe -std=gnu++11 -Wall -Wextra -Wno-unused-parameter -fsigned-char -O2 -Wall -W -D_REENTRANT -fPIC -DCKB_VERSION_STR=\"`cat /home/quillztoxic/Downloads/ckb-next-master/src/ckb/../../VERSION | tr -d \n`\" -DQUAZIP_STATIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/QtWidgets -isystem /usr/include/QtGui -isystem /usr/include/QtNetwork -isystem /usr/include/QtCore -Irelease -isystem /usr/include/libdrm -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o release/kbwidget.o kbwidget.cpp
make[2]: Leaving directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb-wave'
make[1]: Leaving directory '/home/quillztoxic/Downloads/ckb-next-master/src/ckb-wave'
g++ -c -pipe -std=gnu++11 -Wall -Wextra -Wno-unused-parameter -fsigned-char -O2 -Wall -W -D_REENTRANT -fPIC -DCKB_VERSION_STR=\"`cat /home/quillztoxic/Downloads/ckb-next-master/src/ckb/../../VERSION | tr -d \n`\" -DQUAZIP_STATIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/QtWidgets -isystem /usr/include/QtGui -isystem /usr/include/QtNetwork -isystem /usr/include/QtCore -Irelease -isystem /usr/include/libdrm -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o release/colorbutton.o colorbutton.cpp
g++ -c -pipe -std=gnu++11 -Wall -Wextra -Wno-unused-parameter -fsigned-char -O2 -Wall -W -D_REENTRANT -fPIC -DCKB_VERSION_STR=\"`cat /home/quillztoxic/Downloads/ckb-next-master/src/ckb/../../VERSION | tr -d \n`\" -DQUAZIP_STATIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/QtWidgets -isystem /usr/include/QtGui -isystem /usr/include/QtNetwork -isystem /usr/include/QtCore -Irelease -isystem /usr/include/libdrm -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o release/settingswidget.o settingswidget.cpp
g++ -c -pipe -std=gnu++11 -Wall -Wextra -Wno-unused-parameter -fsigned-char -O2 -Wall -W -D_REENTRANT -fPIC -DCKB_VERSION_STR=\"`cat /home/quillztoxic/Downloads/ckb-next-master/src/ckb/../../VERSION | tr -d \n`\" -DQUAZIP_STATIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/QtWidgets -isystem /usr/include/QtGui -isystem /usr/include/QtNetwork -isystem /usr/include/QtCore -Irelease -isystem /usr/include/libdrm -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o release/kblightwidget.o kblightwidget.cpp
g++ -c -pipe -std=gnu++11 -Wall -Wextra -Wno-unused-parameter -fsigned-char -O2 -Wall -W -D_REENTRANT -fPIC -DCKB_VERSION_STR=\"`cat /home/quillztoxic/Downloads/ckb-next-master/src/ckb/../../VERSION | tr -d \n`\" -DQUAZIP_STATIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/QtWidgets -isystem /usr/include/QtGui -isystem /usr/include/QtNetwork -isystem /usr/include/QtCore -Irelease -isystem /usr/include/libdrm -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o release/keymap.o keymap.cpp
g++ -c -pipe -std=gnu++11 -Wall -Wextra -Wno-unused-parameter -fsigned-char -O2 -Wall -W -D_REENTRANT -fPIC -DCKB_VERSION_STR=\"`cat /home/quillztoxic/Downloads/ckb-next-master/src/ckb/../../VERSION | tr -d \n`\" -DQUAZIP_STATIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/QtWidgets -isystem /usr/include/QtGui -isystem /usr/include/QtNetwork -isystem /usr/include/QtCore -Irelease -isystem /usr/include/libdrm -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o release/media_linux.o media_linux.cpp
g++ -c -pipe -std=gnu++11 -Wall -Wextra -Wno-unused-parameter -fsigned-char -O2 -Wall -W -D_REENTRANT -fPIC -DCKB_VERSION_STR=\"`cat /home/quillztoxic/Downloads/ckb-next-master/src/ckb/../../VERSION | tr -d \n`\" -DQUAZIP_STATIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NE [... cut ...]
kbfirmware.h: In constructor ‘KbFirmware::KbFirmware()’:
kbfirmware.h:35:24: warning: ‘KbFirmware::lastFinished’ will be initialized after [-Wreorder]
     quint64 lastCheck, lastFinished;
                        ^~~~~~~~~~~~
kbfirmware.h:28:29: warning:   ‘QNetworkAccessManager* KbFirmware::networkManager’ [-Wreorder]
     QNetworkAccessManager*  networkManager;
                             ^~~~~~~~~~~~~~
kbfirmware.cpp:14:1: warning:   when initialized here [-Wreorder]
 KbFirmware::KbFirmware() :0
 ^~~~~~~~~~

Press enter to exit.
`

light2yellow, 2017-10-28:
Udev is required. It is a part of systemd now. If systemd is not available, you can try eudev.

closed 2017-10-28 by light2yellow

QuillzLouis, 2017-10-28:
@light2yellow I've installed libudev and libudev-devel and I'm getting this error now when building https://hastebin.com/okonehixiz.tex

light2yellow, 2017-10-28:
As you can see, the error hasn't changed. `libudev.h` is not found in the standard search path for headers. `gcc -v -E -` shows the directories where GCC expects to find it.

QuillzLouis, 2017-10-28:
@light2yellow 
I've now got past the libudev error and have got this error:
> Install ckb system-wide [Y/n]? y
> sudo: pam_open_session: System error
> sudo: policy plugin failed session initialization

Full output: https://hastebin.com/gecogewevo.tex

light2yellow, 2017-10-28:
I have no idea how your OS approaches permission granting. `ckb-next` supports Linux and macOS. So, you are pretty much on your own. Patches are welcome.

QuillzLouis, 2017-10-28:
@light2yellow solus is a linux distro.

tatokis, 2017-10-28:
@QuillzLouis is sudo working at all?

tatokis, 2017-10-28:
@QuillzLouis Actually. stupid question. You mentioned you ran sudo ./quickinstall.
I haven't attempted it, but I am sure quickinstall needs to run as a normal user. The script itself asks for root access when it needs it.

light2yellow, 2017-10-29:
> solus is a linux distro

Ah, okay. From the first glance it seemed it's something else. Moreover, I was amused to see it on number 8 on distrowatch...

Anyway, what @tatokis said.

QuillzLouis, 2017-10-29:
@tatokis @light2yellow The output when running ./quickinstall without root 
https://hastebin.com/osavefujad.bash

light2yellow, 2017-10-29:
Well, that is expected. Try
```
sudo make clean
./quickinstall
```