mattanger/ckb-next #322: Corsair Glaiver not working for

MrMcSkittles, 2017-10-29:
Glaiver MAC OS HIGH SIERRA 
Will not read Glaiver upon install from pkg

MatthewMartinFD, 2017-11-10:
I am experiencing the same behavior on the precompiled binary pkg v0.2.8

JarrenJ, 2017-11-10:
I would suggest to attempt building from source and running the QUICKINSTALL script. As it should be updated. I believe the Master branch is the current newest, not sure though haven’t been on as often to check.

MatthewMartinFD, 2017-11-10:
That was going to be my next step, an old post I saw said newdev was the
branch for the glaive, but master looked newer, so i'll try a manual build
of that.

MatthewMartinFD, 2017-11-10:
Not sure if you can provide any quick support, but i did a brew install qt5, and i'm still getting:

Applying required qmake workaround
Error: Cannot apply the workaround. The script assumes Homebrew is installed         and the 'qt' package is installed using it (version 5.9.1).         If you have installed the official Qt package, edit this script and provide proper paths         or install Qt with Homebrew.

MatthewMartinFD, 2017-11-10:
Works now, thanks!

closed 2018-01-21 by tatokis