mattanger/ckb-next #323: Suggestion on a Light Management only version.

owenzhao, 2017-10-29:
My K65 RGB works perfectly with Karabiner-Elements on macOS High Sierra 10.13. F1-F12 Key works the same as a Magic Keyboard. The only existing issue is the light part.

Maybe you can use release a ckb-next just for the light management. I used to use CapsLock key to switch the light of A-Z. Now that is the only needs from ckb-next.

brockbackups, 2017-11-12:
@owenzhao Are you not having modifier key issues, ie... Shift, CTRL being able to select multiple files, folders?

owenzhao, 2017-11-13:
@brockbackups On which app? Just tested on Karabiner-Elements, no issue.

brockbackups, 2017-11-13:
@owenzhao my bad I thought you was talking about ckb-next, sorry for the misunderstanding

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/41

closed 2018-01-22 by tatokis