mattanger/ckb-next #326: [Antergos] ckb-next install fails at qmake-qt5 ("ckb requires at least Qt 5.2!")

tdis8629, 2017-10-31:
# Bug report

- [x] Yes
- [ ] No
## General information

### Source
Tested using ckb-next-latest-git in AUR using pacaur. ckb_next works as expected.

### Branch

Tested using the _master_ branch, in case AUR was the issue, with the same result as above

## Logs & crash reports

Using ./quickinstall:
> ➜  ckb-next git:(master) ✗ ./quickinstall 
> Applying required qmake workaround
> Preparing build files...
> Empty filename passed to function
> Cannot find file: .
> 
> Press enter to exit.

and ./qmake-auto:

> ➜  ckb-next git:(master) ✗ ./qmake-auto 
> Project ERROR: ckb requires at least Qt 5.2!

However, qmake-qt5 -v outputs:

> ➜  ckb-next git:(master) ✗ qmake-qt5 -v   
> QMake version 3.1
> Using Qt version 5.10.0 in /usr/lib

and querying qt5-base results in the following: 

> ➜  ckb-next git:(master) ✗ pacaur -Qi qt5-base 
> Name            : qt5-base
> Version         : 5.10.0beta-1
> Description     : A cross-platform application and UI framework
> Architecture    : x86_64
> ...
> Build Date      : Mon 09 Oct 2017 12:17:03 PM EDT
> Install Date    : Wed 18 Oct 2017 01:48:59 AM EDT
> Install Reason  : Installed as a dependency for another package
> Install Script  : No
> Validated By    : SHA-256 Sum

Attached are the two outputs of quickinstall and qmake auto, with #!/bin/bash -x
[qmake-auto.log](https://github.com/mattanger/ckb-next/files/1431599/qmake-auto.log)
[quickinstall.log](https://github.com/mattanger/ckb-next/files/1431600/quickinstall.log)

## Any other notes & comments?

Installing the stable version of the 0.2.8 package manually and through AUR works as it should.

berolinux, 2017-10-31:
The problem is that the version number is checked in ASCII instead of numbers, so 10 (starts with a 1) is considered less than 9 (starts with a 9).

sed -i -e 's|QT_VERSION, 5.2|QT_VERSION, 5.10|' ckb.pro

gets you going, but isn't the proper fix (though it will do the right thing in just about every case -- ASCII-wise, 5.2 is > 5.10, but 5.1 < 5.10).

tatokis, 2017-10-31:
You can comment out or remove these lines completely and try again.
https://github.com/mattanger/ckb-next/blob/master/ckb.pro#L1-L3

Some kind of workaround needs to be put in place for this, as the cmake pr is being blocked by a few issues.

tatokis, 2017-10-31:
Should be fixed. Thanks!

closed 2017-10-31 by tatokis

tdis8629, 2017-11-01:
I can confirm that it is indeed working on my end. Thank you.

frickler24, 2017-11-03:
Hi @tatokis, I've updated the documentation for the current master branch via travis and got an error for 5.2, because you do this check for versions greater/equal 5.3, not 5.2.
So what is correct?

tatokis, 2017-11-03:
Sorry, my mistake. Corrected it.

commit 5711ba16 (2017-10-31): Correct Qt version check, closes #326