mattanger/ckb-next #327: K70 LUX doesn't work in Linux unless UEFI setup was started before

berolinux, 2017-10-31:
## What is the program's current behavior?

The keyboard works only if the computer has been in UEFI setup mode before booting Linux.
Booting into Linux unattended results in the keyboard not working.
The keyboard isn't set to BIOS mode (and strangely enough still works in UEFI setup).

## What would you like it to do instead?

Make the keyboard work unconditionally.

## General information

### Source
Built from source, most recent commit da28864f

### Branch
master

berolinux, 2017-10-31:
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1431906/ckb-dev-detect-report.gz)

tatokis, 2017-10-31:
This is probably an issue in the USB code that I've been trying to debug for some time, but will probably only be ultimately resolved when we switch to libusb.

Until then, just a wild guess, but try using the 0x20000408 quirk, so that hid-generic initialises the keyboard before ckb-daemon takes over.

berolinux, 2017-11-02:
The 0x20000408 quirk doesn't help (in fact I switched to 0x4 earlier hoping that would fix it). The workaround of going into the UEFI setup tool first doesn't work reliably either (may just have been coincidence that that worked earlier).

Looking at it some more, the keyboard seems to know 3 different states:
1. It works (rarely) -- but when this happens it keeps working until the next reboot (I'm dreading kernel updates these days ;) )
2. It doesn't work and doesn't even show up in lsusb (seems to be most common), with dmesg saying
[Thu Nov  2 20:08:16 2017] usb 3-3: unable to read config index 0 descriptor/all
[Thu Nov  2 20:08:16 2017] usb 3-3: can't read configurations, error -110               
[Thu Nov  2 20:08:16 2017] usb usb3-port3: unable to enumerate USB device
Nevertheless, a non-ckb USB mouse attached to the USB port on the back of the keyboard works.
3. It doesn't work but does show up in lsusb (hasn't happened since I started trying to debug this, but I'm sure I've seen it before - this was the most common state the keyboard was in before I found and installed ckb-next)

berolinux, 2017-11-02:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1678477 seems to be related. Trying a fix based on what they're saying about the Strafe RGB there...

frickler24, 2017-11-03:
@berolinux I have seen some "Caught SIgnal 15" in the report with the hint that ckb-daemon is stopped. Did the messages originate from manually restarting the driver or what sent the signals?

And what kind of logfile entries do you get, if you unplug and replug the keyboard while the driver is running (please also send the kernel messages to the USB interfaces, not only those of ckb).

We obviously have some of these problems with the USB reset again - especially when waking up after standby. That's why I am currently collecting as many points as possible in order to find the reason for it.

berolinux, 2017-11-03:
The problem isn't with ckb-next -- it's with the usbcore initialization code.

Both the keyboard at large and ckb-next on top of it start working perfectly after patching the kernel (tried on 4.13.11 and 4.14-rc7).

Will send that patch to LKML.

[kernel-4.14-K70LUX.patch.txt](https://github.com/mattanger/ckb-next/files/1440573/kernel-4.14-K70LUX.patch.txt)

Probably people will keep coming to ckb-next for a solution - so it might be good to have a copy of the patch somewhere in ckb-next (maybe along with instructions to make it work on other keyboards - I presume the K70LUX isn't the only one with broken USB host code).

berolinux, 2017-11-03:
Patch submitted upstream...
https://lkml.org/lkml/2017/11/3/516

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/42

closed 2018-01-22 by tatokis