mattanger/ckb-next #333: New Corsair STRAFE Non-RGB 1b1c:1b44 (Firmware 1.0)

zeroepoch, 2017-11-07:
I appear to have a newer version of the STRAFE non-rgb keyboard.  It has a different product id than what is listed.  I have an older version of the same keyboard at home which does match the listed product id.  Messing with the usbhid quirks doesn't seem to help with the reliability of the keyboard at boot.  All that helps is disabling USB legacy support which then makes it impossible to easily dual boot etc.  I was hoping ckb-next would help with that but I see it's not detected.  I have attached the file that was requested in the new issue guide.

[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1451319/ckb-dev-detect-report.gz)

zeroepoch, 2017-11-07:
[ckb-daemon.log](https://github.com/mattanger/ckb-next/files/1451439/ckb-daemon.log)

Looking at the attached daemon log from upstart it appears this keyboard is a completely new protocol.  Maybe that's why it's giving me problems while my older STRAFE works reliably with Fedora 25 without having to install ckb or do anything.

tatokis, 2017-11-08:
Please try the latest master. https://github.com/mattanger/ckb-next/commit/4eb96a2c577dab2c4c5886b2eccd6f6be59dbf6f

zeroepoch, 2017-11-08:
I didn't kill the keyboard this time but it also still shows nothing connected in the gui.

zeroepoch, 2017-11-08:
Adding quirks similar to the official quirks in the 4.11 kernel (0x20000400) seems to help the connection and once it starts working it continues until the occasional reboot where I need to reboot again.  I'm not interested in the effects (which is the main purpose of this project) since this is at work I just was hoping to get it to connect more reliably.  I probably won't have the time to debug further, but thanks for making that change so I could try it.

light2yellow, 2017-11-09:
@zeroepoch 

Sorry to say it, but as got unveiled recently in #327, you might want to try to compile your own kernel with the same change as in that issue, but adding the keyboard with the correct product ID. I understand you might not have time to do so, but this is the only way to test the reliability of the setup with this keyboard between reboots before sending any patches upstream, as we can't just send a bunch of IDs and assume it works.

zeroepoch, 2017-11-09:
This looks like a quirk change in usbcore vs the usual usbhid quirk?  Is there anyway to set this from the kernel command line?  If this was at home where I had more time and freedom I'd be happy to test it.  I do have root access on this computer and could install a custom kernel, but I don't have much time right now to experiment with that.

zeroepoch, 2017-11-09:
I believe this may be the same problem as the K70 LUX although it happens only 1/3 of the time now with the quirk on the kernel command line.  Maybe it's not actually helping and just luck.  I don't reboot that often now that everything is setup so I don't expect to deal with this issue much anymore.  Similar to what's mentioned in the link it does seem to work most often when I stop at grub or mess with the BIOS or boot from power off.  I recently disabled CSM (legacy support) to get a proper 4k resolution from UEFI during boot in grub and splash so I'm not sure if that makes a difference or not.

Looking at archived boot logs I got this which appears to be the same issue:

```
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.10.0-38-generic.efi.signed root=UUID=144556df-e613-40f9-b681-3a351fe25502 ro quiet splash usbhid.quirks=0x1b1c:0x1b44:0x20000400 vt.handoff=7
...
[    1.106190] usb 3-13: new full-speed USB device number 2 using xhci_hcd
...
[    6.378415] usb 3-13: unable to read config index 0 descriptor/all
[    6.378418] usb 3-13: can't read configurations, error -110
[    6.498396] usb 3-13: new full-speed USB device number 3 using xhci_hcd
```

Repeats the last 3 lines a bunch of times since it seems to be resetting the device.

Even when the keyboard is working fine and I get no errors and expected dmesg output it still didn't detect the device in ckb-next.  It's a great project and I've played with ckb before the fork with my original STRAFE, but I'm sorry to say I don't have time to play with that either.  In fact I have the lights off since it's always bright in the office.  I bought it mainly for the silent cherry mx keys and typing comfort (muscle memory).  Hopefully you come across someone else with this new model who has more time to help debug.

tatokis, 2017-11-11:
@zeroepoch those logs indicate that the fault definitely from before ckb-daemon can do anything. The quirk is applied to the usb driver itself, not the hid driver like the ones we are used to.

Either way, if you are willing to try one more time, I realised I never properly added it, so just for the sake of supporting this device, can you please check once again with the latest master?
https://github.com/mattanger/ckb-next/commit/9b563545e24df1b95f4c9b847932ac5d7b1f32d1

zeroepoch, 2017-11-13:
Well it does recognize the keyboard now.  It wants to do a firmware upgrade to 2.0.4 like the old STRAFE but ignoring that it kind of works.  I can adjust the brightness but that's about it.  None of the animations seem to do anything.  Since I just keep the backlight off all the time normally I'm not too concerned about that.  Thanks for adding some amount of support to ckb-next for this keyboard.  I'll probably go back to using it normally with the usbhid quirks only unless I get time to try that patch.

closed 2017-11-13 by zeroepoch