mattanger/ckb-next #334: K65 RGB Firmware 3.05

brockbackups, 2017-11-07:
Just updated my firmware on my K65 RGB to 3.05 and CKB-Next can not find the device anymore, it was working with firmware v 2.65

brockbackups, 2017-11-07:
Re-downloaded firmware 1.06 and ckb-next is working again. Anything you need from me to get this working with latest firmware?

tatokis, 2017-11-08:
Please upgrade to the latest firmware again, and run ckb-dev-detect located in the root of this repository.
If you also wish to test something, out of a complete guess on my end, find out which exact model you have and add it to IS_NEW_PROTOCOL() in src/ckb-daemon/usb.h.

brockbackups, 2017-11-09:
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1456270/ckb-dev-detect-report.gz)
Here is the what you requested.

ZirconiumX, 2017-11-11:
`IS_NEW_PROTOCOL()` is [here](https://github.com/mattanger/ckb-next/blob/9300af39869042ba33172978f1b305b47c27368e/src/ckb-daemon/usb.h#L167).

brockbackups, 2017-11-11:
Yeah I finally found it, tried adding IS_K65(kb) to that list and still wouldn't find my device with firmware 3.05

brockbackups, 2017-11-11:
currently using firmware 2.05 and its working

brockbackups, 2017-11-13:
Here is a link to a lot something that might help with this issue https://www.dropbox.com/sh/xfhjo7y67s6gfq1/AAC3qdi3zFQC35fZBgGWl3hJa?dl=0

brockbackups, 2017-11-13:
This is output from ckb-daemon when I load it manually.

`    ckb: Corsair RGB driver v0.2.8
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K65 RGB Gaming Keyboard  at /var/run/ckb1
[W] add_hid (usb_mac.c:799): Got unknown handle (I: 64, O: 2, F: 1)
[E] iterate_devices_hid (usb_mac.c:854): Failed to create device plugin: e00002be
[E] iterate_devices_hid (usb_mac.c:854): Failed to create device plugin: e00002be
[I] Registering EventTap for modifier keys.
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb1
[W] add_hid (usb_mac.c:825): Tried to set up ifhid[3] for device ckb1, but it was already set up. Skipping...
[W] add_hid (usb_mac.c:799): Got unknown handle (I: 64, O: 2, F: 1)`

closed 2018-01-21 by tatokis