mattanger/ckb-next #335: CUE is coming to macOS

mhennessie, 2017-11-08:
This isn't an issue, but CUE is coming to macOS shortly. I am currently running a beta of it and it works great so far. It has all of the features of the Windows variant and maintains the complete look and feel. It is working perfectly with my K65 RGB and M65 Pro RGB while running macOS 10.13.1.

[image]

fleischie, 2017-11-08:
@mhennessie Is there any publicly available information on this topic/beta? Are you referring to http://forum.corsair.com/v3/showthread.php?t=168738 ?

tatokis, 2017-11-08:
Corsair has money to spend for kernel drivers. We do not, so we have to deal with tons of hacks and workarounds.
I'm sorry but this is completely unrelated to the project.

closed 2017-11-08 by tatokis

mhennessie, 2017-11-08:
@fleischie yes that is the topic.

mhennessie, 2017-11-08:
@tatokis how is this unrelated? This project only exist because CUE and drivers for macOS did not. Full support for our devices is coming which is great news.

tatokis, 2017-11-08:
@mhennessie this is a bug tracker for issues related to this project, and this is not an issue.
The software is not exclusive to mac either. This project will still exist after this.

JarrenJ, 2017-11-11:
@mhennessie I have a question for you if you could reply to me here or by email at j.jarren@yahoo.com. I would appreciate it, thanks.

mhennessie, 2017-11-11:
@jarrenj what’s your question?

JarrenJ, 2017-11-11:
@mhennessie so I contacted James from the forum post and he gave me the files I need. However when I try to update to Frimware 3.x.x on my Windows machine, its says it is an invalid Firmware... Could you maybe tell me what I'm doing wrong? I got the firmware .zip for both my devices and when I upload the firmware image it says Invalid... any ideas?

mhennessie, 2017-11-11:
@jarrenj which file are you using? The zipped firmware files I downloaded contained a few files. Also make sure you have downloaded the files for the correct device.

JarrenJ, 2017-11-11:
@mhennessie so I am doign my Scimitar Pro first. I downloaded that .zip and it contained 3 .bin files. I then choose to upload the .zip file to my device through CUE. correct?

mhennessie, 2017-11-11:
@jarrenj you don’t upload the whole zip file. You will upload only one of them. I used the one with APP in the name.

JarrenJ, 2017-11-11:
@mhennessie ok thanks, Ill try that when I get back home. Thanks for the help!