mattanger/ckb-next #337: Issue with System Intermittent Freezes / Lockups after pressing CAPS LOCK Button

Handsomfreddy, 2017-11-10:
Can someone provide a new Mac .pkg file for download with the latest changes to the master branch?  It would be greatly appreciated.

JarrenJ, 2017-11-10:
@Handsomfreddy I found a workaround for he CAPS LOCK button a long time ago. Check for issues with me in it, I think it should pop up. As for the .pkg file. Why not just build from Source by running the ./quickinstall script?

JarrenJ, 2017-11-10:
https://github.com/mattanger/ckb-next/issues/23#issuecomment-275820468 

That is my comment stating the workaround, if that is the same problem you are having.

Handsomfreddy, 2017-11-10:
Thanks Jarren, I’ll give it a go.  Don’t I need xcode to do the build from source?

JarrenJ, 2017-11-10:
I believe you need x-code yes... but I can’t rememeber. It might just be QT that you need.

closed 2018-01-21 by tatokis