mattanger/ckb-next #338: ./quickinstall not working

Nentify, 2017-11-10:
**Source:** git clone
**Branch:** master
**Log:** https://pastebin.com/1gZehhPx

[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1462598/ckb-dev-detect-report.gz)

## What is the program's current behavior?
Error on install

## What is the expected behavior of the program?
Successful installation

## Any other notes & comments?
I ran this command first:

```
sudo apt-get install build-essential libudev-dev qt5-default zlib1g-dev libappindicator-dev
```

```
$ ls -l /usr/lib/x86_64-linux-gnu/qt5/bin/
total 6024
-rwxr-xr-x 1 root root    6344 Jun 28 10:54 fixqt4headers.pl
-rwxr-xr-x 1 root root 1149120 Sep 23 18:13 moc
-rwxr-xr-x 1 root root  284736 Sep 23 18:13 qdbuscpp2xml
-rwxr-xr-x 1 root root   75760 Sep 23 18:13 qdbusxml2cpp
-rwxr-xr-x 1 root root  202736 Sep 23 18:13 qlalr
-rwxr-xr-x 1 root root 2710000 Sep 23 18:13 qmake
-rwxr-xr-x 1 root root 1022144 Sep 23 18:13 rcc
-rwxr-xr-x 1 root root   49240 Jun 28 10:54 syncqt.pl
-rwxr-xr-x 1 root root  649280 Sep 23 18:13 uic
```

ZirconiumX, 2017-11-11:
`/usr/lib/x86_64-linux-gnu/qt5/bin/uic: error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory` points to you having an incorrect Qt5 install. Since I'm an Arch user, I don't know the specific package you need to install; that'd be for @tatokis.

In any case, this isn't a problem with ckb-next.

HenningHeinz, 2017-11-11:
You probably need qtbase5-dev-tools.

tatokis, 2017-11-11:
Can you please post the output of `sudo apt-get install build-essential libudev-dev qt5-default zlib1g-dev libappindicator-dev` ?

Nentify, 2017-11-11:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.4ubuntu1).
libappindicator-dev is already the newest version (12.10.1+17.04.20170215-0ubuntu2).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-0ubuntu2).
qt5-default is already the newest version (5.9.1+dfsg-10ubuntu1).
libudev-dev is already the newest version (234-2ubuntu12.1).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
```

This is a fresh install of Ubuntu 17.10. I'm also running into the same issue when trying to launch VirtualBox, so it's probably not something on your end, but I do appreciate the help.

`qtbase5-dev-tools` is installed.

tatokis, 2017-11-11:
As per https://packages.ubuntu.com/search?searchon=contents&keywords=libQt5Core.so.5&mode=exactfilename&suite=artful&arch=any
try `sudo apt-get install --reinstall libqt5core5a`

Nentify, 2017-11-11:
```
sudo apt-get install --reinstall libqt5core5a
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 to upgrade, 0 to newly install, 1 reinstalled, 0 to remove and 0 not to upgrade.
Need to get 0 B/2,040 kB of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 162361 files and directories currently installed.)
Preparing to unpack .../libqt5core5a_5.9.1+dfsg-10ubuntu1_amd64.deb ...
Unpacking libqt5core5a:amd64 (5.9.1+dfsg-10ubuntu1) over (5.9.1+dfsg-10ubuntu1) ...
Setting up libqt5core5a:amd64 (5.9.1+dfsg-10ubuntu1) ...
Processing triggers for libc-bin (2.26-0ubuntu2) ...
```

```
...
/usr/lib/x86_64-linux-gnu/qt5/bin/uic: error while loading shared libraries: libdouble-conversion.so.1: cannot open shared object file: No such file or directory
make[2]: *** [ui_mainwindow.h] Error 127
make[1]: *** [release] Error 2
make: *** [sub-src-ckb-make_first] Error 2
make: *** Waiting for unfinished jobs....
```

No luck I'm afraid.

Nentify, 2017-11-11:
Running `sudo apt-get install --reinstall liblz4-1 libdouble-conversion1` seemed to fix this error actually. Seems like a lot of packages are broken.

Nentify, 2017-11-11:
Manage to get this far now:

```
In file included from kbfirmware.cpp:1:0:
kbfirmware.h: In constructor ‘KbFirmware::KbFirmware()’:
kbfirmware.h:35:24: warning: ‘KbFirmware::lastFinished’ will be initialized after [-Wreorder]
     quint64 lastCheck, lastFinished;
                        ^~~~~~~~~~~~
kbfirmware.h:28:29: warning:   ‘QNetworkAccessManager* KbFirmware::networkManager’ [-Wreorder]
     QNetworkAccessManager*  networkManager;
                             ^~~~~~~~~~~~~~
kbfirmware.cpp:14:1: warning:   when initialized here [-Wreorder]
 KbFirmware::KbFirmware() :
 ^~~~~~~~~~
/usr/bin/ld: cannot find -lQt5Widgets
/usr/bin/ld: cannot find -lQt5Gui
/usr/bin/ld: cannot find -lQt5Network
collect2: error: ld returned 1 exit status
make[2]: *** [../../bin/ckb] Error 1
make[1]: *** [release] Error 2
make: *** [sub-src-ckb-make_first] Error 2
```

Full log: [log.txt](https://github.com/mattanger/ckb-next/files/1463850/log.txt)

light2yellow, 2017-11-11:
What if you reinstall `qt5-default`?

tatokis, 2017-11-11:
Since it's a dummy package, it won't do much.
Try `sudo apt-get install --reinstall libqt5concurrent5 libqt5core5a libqt5dbus5 libqt5gui5 libqt5network5 libqt5printsupport5 libqt5sql5 libqt5test5 libqt5widgets5 libqt5xml5 libxext-dev qt5-qmake qtbase5-dev-tools qtchooser `

Nentify, 2017-11-11:
Thanks so much for your help!

Got this far and got this error:

```
Finished!
Unapplying qmake workaround
Install ckb system-wide [Y/n]? 
Binaries were installed under /usr/bin
Animations were installed under /usr/lib/ckb-animations

System service: systemd detected
Run the driver as a startup service [Y/n]? 

Start ckb now [Y/n]? 
./quickinstall: line 269:  9023 Aborted                 (core dumped) "$APP" --close > /dev/null 2>&1
```

```
$ ckb
QObject::startTimer: Timers cannot have negative intervals
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Reinstalling the application may fix this problem.
Aborted (core dumped)
```

Ran these 2 commands (just to make sure I catch any/everything that's broken) and tried running ckb again and it's working! Not sure why so many packages seem to be broken, but hopefully it helps someone else having similar difficulties.

```
sudo apt-get install "^libxcb.*" libx11-xcb-dev libglu1-mesa-dev libxrender-dev
sudo apt-get install --reinstall "^libxcb.*" libx11-xcb-dev libglu1-mesa-dev libxrender-dev
```

tatokis, 2017-11-11:
Glad you got it sorted out

closed 2017-11-11 by tatokis