mattanger/ckb-next #341: Mouse unresponsive after suspend/resume cycle

taisph, 2017-11-22:
# Bug report

- [x] Yes
- [ ] No

## General information

### Source

Built from git master ref v0.2.8-224-gb88d8be.

### Branch

master

## Logs & crash reports

Attached.

## What is the program's current behavior?

Mouse is unresponsive after reboot if ckb was running. Animations are still playing on mouse. I have to run `ckb --close` and `systemctl restart ckb-daemon` before mouse can be used again.

## What is the expected behavior of the program?

Mouse is responsive.

## Any other notes & comments?

If `ckb` isn't running when suspending, the mouse works as expected after resuming, so it seems `ckb` may be preventing `ckb-daemon` from handling suspends correctly.

taisph, 2017-11-22:
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1495154/ckb-dev-detect-report.gz)

ZirconiumX, 2017-11-23:
Just as a quick acknowledgement - I have the exact same mouse as you, so I will start investigating.

ZirconiumX, 2017-11-23:
Okay, so with my build of this on b88d8be, I can't actually reproduce this. Come visit us at #ckb-next on Freenode, and mention me on there. I'll be happy to help.

tatokis, 2018-01-21:
https://github.com/mattanger/ckb-next/issues/58

closed 2018-01-21 by tatokis