mattanger/ckb-next #343: Mouse button mapping on Linux (wrong numbers, cannot map to horizontal scroll)

denilsonsa, 2017-11-26:
# Bug report

## General information

* Ubuntu 17.04
* Corsair mouse Harpoon RGB
* Built from git master branch (commit b88d8beb51a16351937558ce5e9c8e155bcdc830)
* [ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1503750/ckb-dev-detect-report.gz)

## Summary

The mouse button numbering in ckb-next interface does not match the actual X11/xinput buttons. In addition, it is impossible to map buttons to horizontal scroll.

## X11 buttons

You can view the button numbers of each click by running `xev -event button` and then clicking on that window.

These are the button numbers I know of:

* `1`, `2`, `3`: Left, middle, right buttons
* `4`, `5`: Scroll up/down
* `6`, `7`: Scroll left/right
* `8`, `9`: Navigation back/forward

## What is the program's current behavior? What's expected?

### Button numbering

At *Special button*, it shows: `4 (Back)`, `5 (Forward)`, `6`, `7`, `8`. However, when testing using `xev`, the actual buttons reported in each mouse click are in fact `8`, `9`, `10`, `11`, `12`. This means **the numbering is off by four**.

I can understand the UI uses these numbers because of cross-platform reasons, or because those numbers are used somewhere in a low-level layer. However, for the end-user, these numbers are confusing, because they do not match whatever is reported by X events.

What do I expect? Either one of these two solutions/behaviors:

* Show the correct number (i.e. 8…12), and only that number.
* Show two numbers for each option. Like this example: `4 (Back) [X11 8]`

I don't know if this applies to other platforms (such as Mac), so maybe this fix should only happen on Linux.

### Horizontal scrolling

The current UI only shows two options for *Wheel*: `Up` and `Down`.

I wanted to remap the side buttons of my mouse to horizontal scrolling. Since I knew the X11 buttons for that (`6`, `7`), I tried remapping them to `6`, `7` in the UI. However, due to aforementioned inconsistency, it did not work (the actual buttons became `10` and `11`).

Then I looked again at the UI, and paid more attention to it, and realized **it is currently impossible to remap buttons to horizontal scroll**.

If the hardware/firmware supports that, I would like to see a way to remap buttons to horizontal scroll in the UI. I don't know if that is possible, I have this mouse for a couple of days, and I haven't even used the CUE tool yet.

### Workaround

I tried to work around it by [remapping the X11 mouse buttons using `xinput`](https://bitbucket.org/denilsonsa/small_scripts/src/default/xmodmap_logitech_mouse_buttons.sh), with a command similar to `xinput set-button-map "INSERT_MOUSE_ID_HERE" 1 2 3 4 5 6 7 6 7`. After that, `xev` correctly reports buttons `6` and `7`, and I could do horizontal scrolling on Gimp. For some reason, it did not work on Chrome, but I haven't investigated nor tried to restart it yet.

## Two issues?

Well, I think these two issues are closely related. That's why I reported them together. If later we figure out they are quite different, we can split them.

ZirconiumX, 2017-11-26:
Okay, so mouse issues are my department (because I'm the only person on the team who has a mouse), but UI issues are for @tatokis.

The mouse buttons are currently [numbered](https://github.com/mattanger/ckb-next/blob/master/src/ckb-daemon/keymap.c#L181-L207) according to the Corsair internal numbering scheme. I have no real idea what that itself is based on, because it makes little sense either way.

It's entirely possible that the issue here is that your Harpoon mouse itself is reporting mouse buttons 8-12, which means we might have to work around it entirely.

As for horizontal scrolling, I think that's going to be a little while off, since we would have to implement that in both the daemon (because it sends the events to the OS) and the GUI (so you can rebind it)/

Either way, thank you for reporting this, and we'll try our best to help.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/46

closed 2018-01-22 by tatokis