mattanger/ckb-next #344: ckb-daemon hangs on exit/shutdown

hexadecatrienoic, 2017-11-26:
Running `ckb-next-git 1:0.2.8.r226.g79987f8-1` from the AUR, on Arch `4.13.12-1-ARCH`.

When ckb-daemon exits, it hangs for around 30 seconds before actually exiting. I have attached the output from `sudo strace ckb-daemon |& tee ckb-daemon-strace.out` [here](https://github.com/mattanger/ckb-next/files/1504313/ckb-daemon-strace.txt). The process was killed by `sudo pkill ckb-daemon` in another terminal, and this was all run with ckb-daemon.service stopped. The exit hangs on line 4120, at `ioctl(3, USBDEVFS_IOCTL` before printing the rest of the line. The rest of the output is printed immediately after it finishes hanging. I can provide more information if required.

ZirconiumX, 2017-11-27:
We know about this issue, though we haven't run `strace` like you have. Since it's hanging in an `ioctl`, that points to the problem being in the kernel, and not in our code.

We plan to switch to libusb on Linux; it might solve it, it might not, but it's worth a shot.

hexadecatrienoic, 2017-11-27:
Ok, I'm happy to help debug if needed.

ZirconiumX, 2017-11-27:
The easiest way to do that is to join us in IRC - #ckb-next on Freenode. That way we get fast response, and it's much easier to do.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/47

closed 2018-01-22 by tatokis