mattanger/ckb-next #345: K70 RGB not being recognized

Frightflight, 2017-11-27:
__Switch on the "Preview" mode for better reading experience.__
Use it to put ticks and then switch back to write in the details where needed.

- [Before you proceed](#before-you-proceed)
- [Devices information](#devices-information)
- [New device support request](#new-device-support-request)
- [Feature request](#feature-request)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior)
    - [What would you like it to do instead?](#what-would-you-like-it-to-do-instead)
    - [Any other notes & comments?](#any-other-notes--comments)
- [Bug report](#bug-report)
    - [General information](#general-information)
        - [Source](#source)
        - [Branch](#branch)
    - [Logs & crash reports](#logs--crash-reports)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior-1)
    - [What is the expected behavior of the program?](#what-is-the-expected-behavior-of-the-program)
    - [Any other notes & comments?](#any-other-notes--comments-1)
- [Something else](#something-else)

---

# Before you proceed

- [ ] __Try__ the latest code by cloning and building the project manually.
- [ ] __Read__ the [`README.md`](https://github.com/mattanger/ckb-next/blob/master/README.md) _in full_.
- [ ] __Skim__ through other `.md` documents in the project's root. They contain more specific information.
- [ ] __Search__ in open _and_ closed [Issues](https://github.com/mattanger/ckb-next/issues). [Here's how](https://help.github.com/articles/searching-issues/).
- [ ] __Avoid__ opening a new issue in case a similar one already exists (don't worry, we get the notifications for closed issues as well, it will be reopened if needed).
- [ ] __Keep__ unrelated issues separate (create a new one if you are sure yours is unique).
- [ ] __Use__ [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to write on GitHub. Use triple backticks if copy-pasting more than _one_ line.

Now, __before you proceed even further__, make sure you are following the recommendations above. It will save a lot of time for you and us in the future.

---

# Devices information

Run [`ckb-dev-detect`](https://github.com/mattanger/ckb-next/blob/master/ckb-dev-detect) from the root of the source code tree:

```console
$ ./ckb-dev-detect
```

Upload `ckb-dev-detect-report.gz` on GitHub with this issue.

---

# New device support request

- [ ] Yes
- [ ] No

<sub><sup>leave empty if you have no new device support requests</sup></sub>

You should upload:
* __Linux__: `ckb-dev-detect-report.gz` is enough
* __macOS__: a full dump following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

- [ ] Yes
- [ ] No

<sub><sup>leave empty if you have no feature requests</sup></sub>

## What is the program's current behavior?
2 Fatal errors detected. CKB.app is openable, but my keyboard is not detected.
## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

- [ ] Yes
- [ ] No

<sub><sup>leave empty if you have no bug reports</sup></sub>

## General information

### Source
(how and where did you get this program, e.g.: _"`ckb-next-git` package in AUR"_ or _"built manually using quickinstall script"_ or _"pkg for macOS"_)
quick install script
### Branch
(fill in if ckb-next was compiled manually, e.g.: _"master"_ or _"newdev"_)

## Logs & crash reports
(any useful information - an OS crash report, debugger's backtrace, any non-standard way of usage (BIOS mode ...)  etc.)

You should upload:
- Linux: `ckb-dev-detect-report.gz` is enough
- macOS:
    * the output of `ckb-daemon`:
    ```console
    $ sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
    $ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ```
    * the output of `ckb` (open another terminal):
    ```console
    $ sudo killall -9 ckb
    $ /Applications/ckb.app/Contents/MacOS/ckb
    ```

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

<sub><sup>if your issue is about something else, try your best to describe it here, otherwise leave empty</sup></sub>

Frightflight, 2017-11-27:
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1504710/ckb-dev-detect-report.gz)

Frightflight, 2017-11-27:
I'm running Mac OS High Sierra 10.13.1
It's a 2016 MacBook Pro with no touch bar

ZirconiumX, 2017-11-27:
High Sierra is at the moment unsupported, because Apple have a tendency of breaking APIs that we rely on. Additionally, none of the dev team actually has a Mac to develop on, so we've got that as an issue as well.

We'll try our best, but don't expect miracles - Apple's developer documentation is awful at the best of times.

Frightflight, 2017-11-27:
Hello!
Thank you for your quick response! This sucks to hear and I'm in no rush, you devs can take your time!
Is there anyway I can make presets for the keyboard on a PC then save it to the keyboard?

Thank you!

ZirconiumX, 2017-11-27:
Sort of - the device itself only supports a single static colour in hardware, so it's not very pretty to look at. But it is possible though CUE for Windows or Mac, or CKB on an older version of Mac.

fleischie, 2017-11-27:
@Frightflight (and everybody else who is interested):

I do have the "older" K70 RGB (non-LUX) and High Sierra available. I can and will check tomorrow after work, whether ckb can detect and interact with it.

Also: What are the 2 fatal errors?

Frightflight, 2017-11-27:
@ZirconiumX  damn. Guess I will have to sit with plain red as my backlighting until something happens. Thank you for the reply!

@fleischie 
[image]

tatokis, 2017-11-27:
@Frightflight please attach the full output, the actual errors are higher up.

fleischie, 2017-11-28:
I can compile the lastest `master` branch successfully (running `qmake && make`) on High Sierra. (There are a lot of warnings, but these are expected, no errors though.) My K70 is detected and works as expected: assign static colors, animations, restore previously saved profiles, custom bindings, custom performance indicators, etc.

Can you make sure, that the daemon is running? Run something like `ps -ef | grep ckb-daemon` and post it's results here, please.

Otherwise this might be an issue with the K70 LUX, which I do not have. Otherwise the keyboard works flawlessly with High Sierra.

Frightflight, 2017-11-29:
./keyaction.h:165:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:166:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
2 warnings generated.
In file included from release/moc_kbwidget.cpp:9:
In file included from release/../kbwidget.h:7:
In file included from ./kb.h:7:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:165:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:166:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
In file included from release/moc_mainwindow.cpp:9:
In file included from release/../mainwindow.h:9:
In file included from ./kbwidget.h:7:
In file included from ./kb.h:7:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:165:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:166:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
2 warnings generated.
In file included from release/moc_kbprofile.cpp:9:
In file included from release/../kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:165:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:166:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
In file included from release/moc_kbmode.cpp:9:
In file included from release/../kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:165:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:166:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
In file included from release/moc_kb.cpp:9:
In file included from release/../kb.h:7:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:165:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:166:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
In file included from release/moc_kbprofiledialog.cpp:9:
In file included from release/../kbprofiledialog.h:5:
[... 101 lines cut ...]
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:165:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:166:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
In file included from release/moc_kbmanager.cpp:9:
In file included from release/../kbmanager.h:8:
In file included from ./kb.h:7:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:165:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:166:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.

Here's what I got after running it again!
@fleischie I'll comment that after this!

Frightflight, 2017-11-29:
@fleischie  here's what I got
ps -ef | grep ckb-daemon
  501 32209 32202   0  7:23PM ttys000    0:00.00 grep ckb-daemon

ZirconiumX, 2017-12-04:
I should probably make an update on this, so:

The output log you posted doesn't actually contain any fatal errors, so it's not very helpful to us.

You have the ckb-daemon running, along with the GUI, but you're not getting your keyboard detected. This is odd, so I suggest you join us [in IRC](http://webchat.freenode.net/?channels=%23ckb-next&uio=d4). Make sure to mention "ZirconiumX" in your post.

Frightflight, 2017-12-04:
Hi! I don't know if this is any help, but I did run the detect script and have the following:
I'm sorry I'm a total noob at this! I was expecting a plug/play software :)
@ZirconiumX  I can join you all in IRC in a couple of days! 
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1527927/ckb-dev-detect-report.gz)

ZirconiumX, 2017-12-04:
>I was expecting a plug/play software :)

80% of the time it is, or it will be after a little niggle ("It doesn't work!" "Have you started the daemon?" "Oh, that fixed it, thanks!"). We'll be able to investigate more when we get chance to talk to you.

fleischie, 2017-12-04:
Sorry I wasn't following up on this until now: The ckb daemon is not running (only the grep command is in the process list).

Please run `sudo launchctl load -w /Library/LaunchDaemons/com.ckb.daemon.plist` or something similar (use tab completion in the terminal to find the correct entry in the directory). This should automatically start the daemon on

huttj, 2017-12-10:
I had the same issue on Sierra 10.12.6. I tried the prebuilt binary and building from master.

In the end, it was resolved with a system restart. The program works perfectly, now.

I ran `sudo launchctl load -w /Library/LaunchDaemons/com.ckb.daemon.plist` before I rebooted.

closed 2018-01-21 by tatokis