mattanger/ckb-next #346: K55 IS NOT RECONIZE

mattzelda, 2017-11-27:
please , can someone do a version of CKB for the k55 , I really need it so if someone can do this, it will be be so great !

jmcgraw159, 2017-11-30:
Second this

ZirconiumX, 2017-11-30:
You'll probably notice we ignored this issue, unlike most of the other ones here.

Nobody with a K55 has stepped up to IRC to help us out with it. 

Even if they did, the functionality of the K55 is severely limited by hardware.

We don't think it's worth our time at the moment, and because we're not getting paid for any of this, we have the right to pick and choose which devices to work on; for example we recently got the K95 Platinum working, and we're currently working on Corsair firmware version 3.

If you want the K55 supported so much, we have an IRC channel. Join us.

tatokis, 2017-12-01:
Closing in favour of #277.

closed 2017-12-01 by tatokis

OvermindDL1, 2017-12-01:
I.E., they need someone with a K55 to join IRC so they can walk that person through steps on dumping the USB packets, usually done by running Windows in a VM and dumping the USB packets as they pass between the VM with the official drivers installed and the OS.  :-)

CoroIa, 2020-01-12:
i have a k55 and currently built a version that runs and saves profiles (outside of Demo Mode) on a dual booted Ubuntu/Chrome OS Chromebook. lol quite an odd pair of OS and type of "computer" to attempt running this on.. if you would like i can share pictures of the build process and information on getting the damn thing out of "demo mode" as that very thing had me stumped for a few months.. with no active help i made a break through. again i built this on a chromebook which was deemed impossible but maybe there wouldnt be too many bugs along the way attempting to troubleshoot your system and OS... Nothing is impossible my friend, if they say it is then it only means it hasnt been done yet and we will be the first to do it which is pretty exciting i think..

tdis8629, 2020-01-12:
Please view ckb-next/ckb-next. This specific fork is no longer being
maintained.