mattanger/ckb-next #348: Move ckb-animations to /usr/libexec

mrueg, 2017-11-27:
# Bug report

- [x] Yes
- [ ] No

## General information
ckb-animations were installed to /usr/bin, now they are installed to /usr/lib.
https://github.com/mattanger/ckb-next/commit/f347e60df211c60452f95084b6c46dc4ec5f42ee#diff-a638147988397087db799b663c7a6616

## What is the program's current behavior?
Installs ckb-animations to /usr/lib
## What is the expected behavior of the program?
Install ckb-animations to /usr/libexec (this seems to be a more natural place and avoids issues with installing it into a hardcoded /usr/lib which could include 64 bit (then it should be installed to /usr/lib64) or 32bit libraries on some systems.

## Any other notes & comments?
See: https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch04s07.html

tatokis, 2017-11-28:
They will be moved with the switch to cmake. See https://github.com/mattanger/ckb-next/pull/281

ZirconiumX, 2017-11-30:
The problem is that putting them in `/usr/libexec` conflicts with the Arch Linux packaging guidelines, which say to put them in `/usr/lib/ckb-next/`, so we're at something of an impasse here.

Maybe we need a configuration variable for where the animations are.

light2yellow, 2017-12-01:
I will quote myself from #281:

> See wiki for the whole list of options as it contains some important ones. For example, ckb-next follows GNU Coding Standards and by default the path to animations is ${CMAKE_INSTALL_PREFIX}/${CMAKE_INSTALL_LIBEXECDIR}/ckb-next-animations which results into /usr/local/libexec/ckb-next-animations by default.

This means that every distro can set the path according to their guidelines, so, it won't cause any issues. I guess it should be kept open until the merge.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/48

closed 2018-01-22 by tatokis