mattanger/ckb-next #349: ckb-next for K70 RGB not working properly in ubuntu 17.10

benhur10cp, 2017-11-28:
Hit "change color" button causes the program to stop working.
I'm on motherboard msi b250m pro vh.

[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1511883/ckb-dev-detect-report.gz)

Thanks a lot.

light2yellow, 2017-11-29:
See https://bbs.archlinux.org/viewtopic.php?id=214147.

closed 2017-11-29 by light2yellow

tatokis, 2017-11-29:
https://github.com/mattanger/ckb-next/issues/175#issuecomment-299453783 with instructions for Ubuntu

benhur10cp, 2017-11-29:
@light2yellow i understood, but i'm not able to do it. thanks.

light2yellow, 2017-12-01:
@benhur10cp 
Okay, we can walk through it together. What have you tried? The thread describes a plethora of solutions with some of them working for some setups.

light2yellow, 2017-12-01:
@benhur10cp 
Just so that you understand - it's not because I want to waste your time while you read 5 pages of some forum, but because there is no single solution to it - I myself have had to apply two different approaches on two almost identical Arch Linux installations, when this problem was still a problem for Arch Linux. And as you can see on that thread, people are reporting mixed level of success. You can also google for your specific distro, but the thread above is the most comprehensive I have ever seen on this particular issue.

benhur10cp, 2017-12-02:
So, what I did, I installed the qt5ct from [(https://launchpad.net/~mati75/+archive/ubuntu/qt5ct)], restart the pc, opened the settings of qt5, set the style for gtk2 and apply. For me it worked perfectly.
Thank you very much.