mattanger/ckb-next #350: Strafe RGB No longer recognised

SG420, 2017-11-30:
__Switch on the "Preview" mode for better reading experience.__
Use it to put ticks and then switch back to write in the details where needed.

- [Before you proceed](#before-you-proceed)
- [Devices information](#devices-information)
- [New device support request](#new-device-support-request)
- [Feature request](#feature-request)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior)
    - [What would you like it to do instead?](#what-would-you-like-it-to-do-instead)
    - [Any other notes & comments?](#any-other-notes--comments)
- [Bug report](#bug-report)
    - [General information](#general-information)
        - [Source](#source)
        - [Branch](#branch)
    - [Logs & crash reports](#logs--crash-reports)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior-1)
    - [What is the expected behavior of the program?](#what-is-the-expected-behavior-of-the-program)
    - [Any other notes & comments?](#any-other-notes--comments-1)
- [Something else](#something-else)

---

# Before you proceed

- [ x] __Try__ the latest code by cloning and building the project manually.
- [x ] __Read__ the [`README.md`](https://github.com/mattanger/ckb-next/blob/master/README.md) _in full_.
- [ x] __Skim__ through other `.md` documents in the project's root. They contain more specific information.
- [ x] __Search__ in open _and_ closed [Issues](https://github.com/mattanger/ckb-next/issues). [Here's how](https://help.github.com/articles/searching-issues/).
- [x ] __Avoid__ opening a new issue in case a similar one already exists (don't worry, we get the notifications for closed issues as well, it will be reopened if needed).
- [ x] __Keep__ unrelated issues separate (create a new one if you are sure yours is unique).
- [ x] __Use__ [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to write on GitHub. Use triple backticks if copy-pasting more than _one_ line.

Now, __before you proceed even further__, make sure you are following the recommendations above. It will save a lot of time for you and us in the future.

---

# Devices information

Run [`ckb-dev-detect`](https://github.com/mattanger/ckb-next/blob/master/ckb-dev-detect) from the root of the source code tree:

```console
$ ./ckb-dev-detect
```

Upload `ckb-dev-detect-report.gz` on GitHub with this issue.

---

# New device support request

- [ ] Yes
- [x

 ] No

<sub><sup>leave empty if you have no new device support requests</sup></sub>

You should upload:
* __Linux__: `ckb-dev-detect-report.gz` is enough
* __macOS__: a full dump following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

- [ ] Yes
- [ ] No

<sub><sup>leave empty if you have no feature requests</sup></sub>

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

- [ ] Yes
- [ ] No

<sub><sup>leave empty if you have no bug reports</sup></sub>

## General information

### Source
(how and where did you get this program, e.g.: _"`ckb-next-git` package in AUR"_ or _"built manually using quickinstall script"_ or _"pkg for macOS"_)

### Branch
(fill in if ckb-next was compiled manually, e.g.: _"master"_ or _"newdev"_)

## Logs & crash reports
(any useful information - an OS crash report, debugger's backtrace, any non-standard way of usage (BIOS mode ...)  etc.)

You should upload:
- Linux: `ckb-dev-detect-report.gz` is enough
- macOS:
    * the output of `ckb-daemon`:
    ```console
    $ sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
    $ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ```
    * the output of `ckb` (open another terminal):
    ```console
    $ sudo killall -9 ckb
    $ /Applications/ckb.app/Contents/MacOS/ckb
    ```

## What is the program's current behavior?
STRAFE RGB is not detected by CKB, it only says No devices connected. I have tried on both my laptop and desktop, both running Manjaro. this issue seems to have only appeared after an update. It works fine in Windows 8.1 Embedded Industry Pro with the proprietary CUE driver installed. I have tried all 4 AUR packages and they all have the same behaviour.
---

# Something else

<sub><sup>if your issue is about something else, try your best to describe it here, otherwise leave empty</sup></sub>
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1516755/ckb-dev-detect-report.gz)

ZirconiumX, 2017-11-30:
Did you run `systemctl start ckb-daemon`? It doesn't run by default, because of the Arch packaging guidelines.

Nepoxx, 2017-11-30:
@ZirconiumX that fixed the "issue" for me, I'd suggest adding that information on the README :)

SG420, 2017-11-30:
@ZirconiumX That fixed the issue, thanks.

closed 2017-11-30 by SG420

ZirconiumX, 2017-11-30:
It's in the AUR package:

```sh
_as_service() {
    cat << EOF

You might want to run ckb-daemon as a systemd service:

    systemctl enable ckb-daemon.service
    systemctl start ckb-daemon.service

EOF
}
```

light2yellow, 2017-12-01:
@Nepoxx 
> I'd suggest adding that information on the README

I'd suggest reading the output of `pacman`. It displays the message above every time you install and/or update a package. Regarding the rationale, see https://wiki.archlinux.org/index.php/Systemd#Enable_installed_units_by_default. I am not very happy about this, too and even started/enabled the daemon some time ago by default, but people told me it's not something they want and not something Arch Linux packaging guidelines adore.

Nepoxx, 2017-12-01:
I'm not arguing whether Arch or the Arch way is right or not, I'm just saying that adding this to the readme could prevent issues like these. It's easy to miss the output of pacman/pacaur/yaourt/etc, especially when installing multiple packages.

The UI could also mention that it can't communicate with the daemon? I could PR that, thoughts?

ZirconiumX, 2017-12-01:
It does. There's a "driver inactive" message, which @fleischie is working on making more obvious.

fleischie, 2017-12-02:
Yeah, I guess the Readme is at a length, where nobody even bothers to read it for troubleshooting.

@Nepoxx check out my PR #354. If it works, it should open a dialog to inform you about the inactive daemon. 😄