mattanger/ckb-next #352: Is this still the correct place to get ckb-next?

mjones153, 2017-11-30:
Is this still the correct place to get ckb-next? 

After looking at https://groups.google.com/forum/#!topic/ckb-next/L-E8RiKxPFk  and seeing no further update to that post, I figured checking would be best.

Thank you for your time.

OvermindDL1, 2017-11-30:
Yep, it is quite active here, look at the recent commits as one example.  :-)

ZirconiumX, 2017-11-30:
Yes, this is still the correct place to get ckb-next. We plan to wait for the domain to expire to see if *that* gets @mattanger's attention. But for now, we're not going anywhere.

OvermindDL1, 2017-11-30:
Well if you do set up a new site I have servers available if you don't want just a simple github-pages static site.

ZirconiumX, 2017-11-30:
Thank you, but no thank you. If we gave you the website, then you disappeared, we'd be in the same situation.

tatokis, 2017-12-01:
@mjones153 That is mostly used for announcements, next one will probably be for our next release (which is going to be a big one). Either way, as it has been mentioned, we are still here for the moment, but even if we were to move, we would've made it as obvious as possible.

@OvermindDL1 Thanks, it is much appreciated, however as @ZirconiumX explained, it is possible that we would end up in the same situation again.

closed 2017-12-01 by tatokis

OvermindDL1, 2017-12-01:
> @OvermindDL1 Thanks, it is much appreciated, however as @ZirconiumX explained, it is possible that we would end up in the same situation again.

Understandable yep, though I do have a bit of a record, my servers have been hosting a variety of open source and community sites since 2000.  :-)

mjones153, 2017-12-01:
Thank you very much.