mattanger/ckb-next #353: K70 RGB Lux Product ID

TheFinestSurgeon, 2017-12-01:
The K70 RGB Lux Product ID is different from the other K70 SKU's with the value being "0x1b33" and should be added to the README.md as "usbhid.quirks=0x1B1C:0x1B33:0x20000408"

Im not entirely sure that this is the right way to report such a thing feel free to correct me otherwise
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1521149/ckb-dev-detect-report.gz)

ZirconiumX, 2017-12-01:
`1b33` is [already in the daemon](https://github.com/mattanger/ckb-next/blob/master/src/ckb-daemon/usb.h#L67). I personally don't think it's necessary to keep copy/pasting the same section over and over again. Maybe I'll write a script to create them for the user.

light2yellow, 2017-12-01:
@ZirconiumX I think we should add the IDs to the wiki page with supported devices (make a table out of that list) and, instead of enumerating them every time in the readme, just explain people how to construct the quirks - basically `0x1b1c:PID:[0x20000408|0x4]`, where PID can be received by executing
```bash
lsusb | grep 1b1c | awk '{ print $6 }' | awk '{ split($0, a, ":"); print a[2] }'
```
(maybe not the nicest way to do this, but it does its job)

light2yellow, 2018-01-01:
The modified version of the above landed in the wiki page ([Troubleshooting](https://github.com/mattanger/ckb-next/wiki/Troubleshooting)). Wanted to do it long time ago, thanks, @TheFinestSurgeon & @ZirconiumX.

closed 2018-01-01 by light2yellow