mattanger/ckb-next #355: Is there a way to verify the download?

mjones153, 2017-12-02:
Is there a way to verify the download with a checksum (such as sha512 or sha256)?  And is there a way to verify the signature of the checksum with gpg?

Thank you.

ZirconiumX, 2017-12-02:
We just use git, so none of our downloads are signed, other than the internal verification that git itself uses.

mjones153, 2017-12-02:
Ok, when I do download it, is there a way to verify that the download file succeeded with just a regular checksum? Thank you.

ZirconiumX, 2017-12-02:
No. Only using the Git checksum.

tatokis, 2017-12-02:
@mjones153 Unfortunately no.
If you use git to clone the repository, it will either be done through ssh or https, both of which are encrypted and prevent tampering between you and github.

Adding signed hashes on the releases is a good idea for the future, but currently our last release is outdated, so you might as well get a fresh copy from master.

If you really do desire, this is the sha512sum for the 0.2.8 release tarball, signed with the ckb-next PGP key that can be found in this repository.
```
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

6159a1abd569e4c31cd8f4ba49232993078d92173a52f91d392bc27e30af62c1a80e953a0661b66bb87bd2b24625a5a6fa68c60140c12851ac96cb19df4ee960  028.tar.gz
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJaIx+jAAoJEN/iSOvc7y9flGEH/jvQcLYV+T+VzH9V1iyLtsAz
x9fdW9T0SHzEAaVtFWB6Y+aggj0nBuw717zwd05l8e2p86bH+j7Y1/xgLoB+hsoK
wHpmU+wPYOlO5Fu/0Ips+bBfrE719qbE608saho/9TCzVLNL0oiXV00vBEldWp1L
HLg/qO4E7m90Ul7zwfBE5bXNwc4m69sEdgy2R8Qpgz2Zt8R3k+Bhuqi25loUba9q
1NmXQCbzCXEWv9OW2pLKmBd6ac1IHswBAJFjTI876JEVb2N7/79SkwPPi/2/xrhp
1mbSg4fMHT4t2kXEsGBX/tj5Y4DKWFJsrytozC+qs60yAoUhEHpwLnmmKWotKvA=
=AOJZ
-----END PGP SIGNATURE-----
```

Alternatively, I could upload a tar archive of the current master somewhere and sign the hash for that, but this is going to be more of a one shot.

tatokis, 2017-12-02:
Looks like github didn't like that, hold on.

tatokis, 2017-12-02:
Attached the signed hash. 
[sha512sum.txt](https://github.com/mattanger/ckb-next/files/1524241/sha512sum.txt)

mjones153, 2017-12-03:
One thing I should ask, what is the sha512 or sha256 sum for the current master.zip download?  Just as a way to double check on my end that the downloaded zip file has the right checksum.

light2yellow, 2017-12-03:
> If you really do desire, this is the sha512sum for the 0.2.8 release tarball, signed with the ckb-next PGP key that can be found in this repository.

This is incorrect. The release was signed with my own [old] key, which is now revoked.

> One thing I should ask, what is the sha512 or sha256 sum for the current master.zip download?

sha256 is 772ee64e0dac094310f560dbf0b0ef274d531766415318f3c60f5beeaed0f8b8 as of e842da0183ed9fc757ad31f59d1ea776192559f2. But as @tatokis said, the Ss in HTTPS and SSH mean "secure", therefore, unless your whole system has already been compromised, a MITM cannot happen.

closed 2017-12-03 by tatokis

light2yellow, 2017-12-03:
> If you really do desire, this is the sha512sum for the 0.2.8 release tarball, signed with the ckb-next PGP key that can be found in this repository.

Just to omit any confusion: in their message @tatokis meant a hand-signed archive and not the git tag in git history, so, what I said is irrelevant.

mjones153, 2017-12-03:
Ah that makes sense.  With the SSH and HTTPS, is that only specific to cloning the repository, or does that apply to downloading the zip as well?

ZirconiumX, 2017-12-03:
For the zip, too.