mattanger/ckb-next #356: ckb-next 2.8.0 No devices connected for Corsair Strafe No RGB Mac Os X El Capitan

Mazhaa, 2017-12-03:
I have the Corsair Strafe no Rgb keyboard and ckb-next 2.8.0 show me "No devices connected" when I plug my keyboard . Can someone help me ?

light2yellow, 2017-12-03:
Output of `launchctl list | grep ckb-daemon`?

Mazhaa, 2017-12-03:
-	0	com.ckb.ckb
1293	0	com.apple.xpc.launchd.oneshot.0x1000000b.ckb
There are this two things when I type launchctl list on terminal

light2yellow, 2017-12-27:
Sorry, I forgot to answer.

First of all, it looks like you did `launchctl list | grep ckb` instead of the above. Secondly, [newer] Strafe non-RGB was added after 0.2.8, so build the program from source. Thirdly, I probably forgot to mention `sudo`, I don't remember whether `launchd` requires sudo to view `root` jobs. So, in case it still isn't detected after being built manually, execute `sudo launchctl list | grep ckb-daemon` (and post it here). In case there will be no output, execute `sudo launchctl load -w /Library/LaunchDaemons/com.ckb.daemon.plist`.

monsonjeremy, 2018-01-04:
I've noticed that this issue happens after restarting my macbook. It works fine when i reinstall CKB but as soon as i perform a restart i lose it :(

Ricardonacif, 2018-01-07:
For me when I restart the computer it doesn't work. What I have to do to make it work is: kill the unplug keyboard, kill ckb-daemon process, open ckb appllication, then plug keyboard again. I believe it has something to do with starting the MacOS with the keyboard plugged in.

light2yellow, 2018-01-07:
Anyways, with the default setup after quickinstall I cannot reproduce on hackintosh on 10.13.whatever.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/49

closed 2018-01-22 by tatokis