mattanger/ckb-next #357: [Bug Report] Daemon - Dota 2 - Text Box de-focuses when typing the letter 'h'

WtGuy, 2017-12-03:
- [Devices information](#devices-information)
- [Bug report](#bug-report)
    - [General information](#general-information)
        - [Source](#source)
        - [Branch](#branch)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior-1)
    - [What is the expected behavior of the program?](#what-is-the-expected-behavior-of-the-program)
    - [Any other notes & comments?](#any-other-notes--comments-1)

---
# Devices information
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1525140/ckb-dev-detect-report.gz)

---

# Bug report

- [x] Yes
- [ ] No

## General information

### Source
My program was acquired using the Arch Linux Aur package ckb-next-git. Just in case, I also tried ckb-next from the aur (standard) and downloading from source and compiling. All results yield the same issue.

### Branch
master on the time I tried from source.

## Logs & crash reports
N/A

## What is the program's current behavior?
This is a bug specific to Dota 2 which has been described in detail [here](https://github.com/ValveSoftware/Dota-2/issues/1299). Essentially, using the letter 'h' in any text box instance (minus the console) causes the focused box to become de-focused. This only happens when the ckb-next-daemon is running and from the linked issue, is not an isolated case.

## What is the expected behavior of the program?
Text boxes should function as normal and type the letter 'h'

## Any other notes & comments?
Please view the attached issue linked above for more details. I will try and get other users to also submit their information in this issue to hopefully determine the root cause. Note that I initially thought this was a bug with Dota 2, but with other users coming out and having the same issue specifically using ckb-next, I figured it may be beneficial to submit a bug report here

light2yellow, 2017-12-03:
Does the issue happen when you don't use ckb-next?
Execute
```
sudo systemctl stop ckb-daemon
```
and then launch the game.

ZirconiumX, 2017-12-04:
@l2y: `This only happens when the ckb-next-daemon is running`

ZirconiumX, 2017-12-04:
@WtGuy We'll investigate, but just as a start, can you compare what `xev` prints for typing the `h` key with and without ckb-daemon running?

tatokis, 2017-12-04:
@WtGuy Can you please download, compile, and run sdl2-jstest with ckb-daemon running?

```
./sdl2-jstest -l
```
You can find it here https://github.com/Grumbel/sdl-jstest

tatokis, 2017-12-04:
I should add that I am unable to reproduce this bug on my machine, running Ubuntu 16.04.

WtGuy, 2017-12-05:
First of all, a great thank you for the fast response. I know this is free open source software and sometimes, these things can take a bit.

@ZirconiumX I've attached two txt files associating to both the daemon running and not running.
- [xev run with daemon not running](https://github.com/mattanger/ckb-next/files/1529552/hKeyDaemonNotRunning.txt)
- [xev run with daemon running](https://github.com/mattanger/ckb-next/files/1529553/hKeyDaemonRunning.txt)

@tatokis I've attached the txt report from the run and have tried the ckb software without my controller plugged in as well with the same results. Also, as a side note, I believe someone in the other issue posted had said they were having the issue with Ubuntu 17.10, so it may be a newer package problem.
- [sdl2-jstest command output with daemon running](https://github.com/mattanger/ckb-next/files/1529554/sdl2Info.txt)

KuriKai, 2017-12-05:
Here are the results of running the command on my computer

Operating System: Ubuntu 17.10
Kernel: Linux 4.14.1-041401-generic
Architecture: x86-64

[sdl2-jstest.txt](https://github.com/mattanger/ckb-next/files/1530498/sdl2-jstest.txt)
[xev-with-ckb-next-daemon.txt](https://github.com/mattanger/ckb-next/files/1530499/xev-with-ckb-next-daemon.txt)
[xev-without-ckb-next-daemon.txt](https://github.com/mattanger/ckb-next/files/1530500/xev-without-ckb-next-daemon.txt)

ZirconiumX, 2017-12-05:
Progress update: We think this is actually an issue in libinput. @tatokis can't reproduce on Ubuntu 16.04, because that version of Ubuntu uses evdev for input, while Ubuntu 17.10 and Arch both use libinput. 
After firing up a VM with 17.10, we can now reproduce. Either way, the problem isn't actually SDL's fault.

Can we get the topic renamed please? How about "libinput detects ckb devices as joysticks"?

tatokis, 2017-12-05:
Quick update, turns out it's newer udev versions marking our virtual devices as a joysticks.
Have been looking for a way of specifying the device type in uinput, however I can't seem to find anything so far.

In the meantime, this can be confirmed by running 
```
udevadm info -p /sys/devices/virtual/input/inputXX/eventYY | grep ID_INPUT_
```
Where XX and YY are the appropriate IDs.

On my computer, there is no JOYSTICK entry:
```
$ udevadm info -p /sys/devices/virtual/input/input35/event22 | grep ID_INPUT_
E: ID_INPUT_KEY=1
E: ID_INPUT_KEYBOARD=1
E: ID_INPUT_MOUSE=1
```

WtGuy, 2017-12-05:
Updated title and put a little comment in the issue information to list its former description as to not confuse new people coming over from the other thread

ZirconiumX, 2017-12-06:
Our current solution is to create `udev` rules to remove these attributes. It's a hack, but oh well.

tatokis, 2017-12-07:
We have a udev rule that appears to work. It would be appreciated if you could test it.
```
ACTION=="add|change", SUBSYSTEM=="input", ATTRS{name}=="ckb[0-9]: [A-Za-z0-9]*", ENV{ID_INPUT_JOYSTICK}=""
```
Install it in `/etc/udev/rules.d/99-ckb-daemon.rules` and restart ckb-daemon.

This should hopefully fix the issue.

fgblomqvist, 2017-12-07:
Just tried it and it works, kudos to you! :D

KuriKai, 2017-12-08:
This also works for me, thanks!

Baguettedood, 2017-12-08:
The udev rule fixes my issues with Rocket League which I mentioned in the IRC a while ago.
It looks like the issue arose in (what is most likely the udev rules included in) systemd 234, and remains in systemd 235.

WtGuy, 2017-12-08:
Good to go here as well. Thank you for the fast resolution.

On a side note: I didn't close this issue on this comment in case you guys wanted to keep it open as a reference to potentially provide a more permanent fix in the future. If not needed, don't hesitate to close the issue.

ZirconiumX, 2017-12-10:
@tatokis Should we create a PR for this and merge it?

tatokis, 2017-12-10:
@ZirconiumX Sure.
Find the right place to put it in the repo, you can commit the udev rule and I can modify quickinstall if you're okay with that.
We should poke the maintainers that don't use quickinstall and tell them to manually install the rule too.

commit 27a2ea9d (2017-12-16): Add udev rule, fixes #357

Thanks to @ZirconiumX for writing the rule.
commit 142b307e (2017-12-17): Add udev rule, fixes #357

Squashed commit of the following:

commit 27a2ea9dda9f8ab097252ac71644e423177d7562
Author: Tasos Sahanidis <tasos@tasossah.com>
Date:   Sat Dec 16 18:49:41 2017 +0200

    Add udev rule, fixes #357

    Thanks to @ZirconiumX for writing the rule.

Closes #365.
closed 2017-12-17 by light2yellow