mattanger/ckb-next #358: Linux encrypted HD

vonschutter, 2017-12-05:
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1530610/ckb-dev-detect-report.gz)
__Switch on the "Preview" mode for better reading experience.__
Use it to put ticks and then switch back to write in the details where needed.

- [Before you proceed](#before-you-proceed)
- [Devices information](#devices-information)
- [New device support request](#new-device-support-request)
- [Feature request](#feature-request)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior)
    - [What would you like it to do instead?](#what-would-you-like-it-to-do-instead)
    - [Any other notes & comments?](#any-other-notes--comments)
- [Bug report](#bug-report)
    - [General information](#general-information)
        - [Source](#source)
        - [Branch](#branch)
    - [Logs & crash reports](#logs--crash-reports)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior-1)
    - [What is the expected behavior of the program?](#what-is-the-expected-behavior-of-the-program)
    - [Any other notes & comments?](#any-other-notes--comments-1)
- [Something else](#something-else)

---

# Before you proceed

- [X ] __Try__ the latest code by cloning and building the project manually.
- [ X] __Read__ the [`README.md`](https://github.com/mattanger/ckb-next/blob/master/README.md) _in full_.
- [ ] __Skim__ through other `.md` documents in the project's root. They contain more specific information.
- [ ] __Search__ in open _and_ closed [Issues](https://github.com/mattanger/ckb-next/issues). [Here's how](https://help.github.com/articles/searching-issues/).
- [ ] __Avoid__ opening a new issue in case a similar one already exists (don't worry, we get the notifications for closed issues as well, it will be reopened if needed).
- [ ] __Keep__ unrelated issues separate (create a new one if you are sure yours is unique).
- [ ] __Use__ [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to write on GitHub. Use triple backticks if copy-pasting more than _one_ line.

Now, __before you proceed even further__, make sure you are following the recommendations above. It will save a lot of time for you and us in the future.

---

# Devices information

Run [`ckb-dev-detect`](https://github.com/mattanger/ckb-next/blob/master/ckb-dev-detect) from the root of the source code tree:

```console
$ ./ckb-dev-detect
```

Upload `ckb-dev-detect-report.gz` on GitHub with this issue.

---

# New device support request

- [ ] Yes
- [ ] No

<sub><sup>leave empty if you have no new device support requests</sup></sub>

You should upload:
* __Linux__: `ckb-dev-detect-report.gz` is enough
* __macOS__: a full dump following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

- [X ] Yes
- [ ] No

<sub><sup>Please add support for K70 RGB/Ubuntu with encrypted hard drive</sup></sub>

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

- [ ] Yes
- [ ] No

<sub><sup>leave empty if you have no bug reports</sup></sub>

## General information

### Source
(how and where did you get this program, e.g.: _"`ckb-next-git` package in AUR"_ or _"built manually using quickinstall script"_ or _"pkg for macOS"_)

### Branch
(fill in if ckb-next was compiled manually, e.g.: _"master"_ or _"newdev"_)

## Logs & crash reports
(any useful information - an OS crash report, debugger's backtrace, any non-standard way of usage (BIOS mode ...)  etc.)

You should upload:
- Linux: `ckb-dev-detect-report.gz` is enough
- macOS:
    * the output of `ckb-daemon`:
    ```console
    $ sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
    $ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ```
    * the output of `ckb` (open another terminal):
    ```console
    $ sudo killall -9 ckb
    $ /Applications/ckb.app/Contents/MacOS/ckb
    ```

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

<sub><sup>if your issue is about something else, try your best to describe it here, otherwise leave empty</sup></sub>

ZirconiumX, 2017-12-05:
@tatokis Can we close this as invalid? I don't actually know what's being asked here.

light2yellow, 2017-12-05:
It says
> Please add support for K70 RGB/Ubuntu with encrypted hard drive

Took me a while, though.

@vonschutter 
1. Learn to use Markdown (as "Before you proceed" section suggests).
2. Your report mentions VOID headset and does not mention K70. You did exactly the opposite. I am completely confused.
3. VOID headset is not supported and hardly ever be.
4. K70s are supported.
5. I don't think there can be any problems with disk encryption and ckb-next. I am completely confused for the second time.

vonschutter, 2017-12-06:
Hi,

in the readme.md is says, and I can confirm that it does not work if you have an encrypted hard drive:

If it still doesn't work, try replacing `0x20000408` with `0x4`. Note that this will cause the kernel driver to ignore the device(s) completely, so you need to ensure ckb-daemon is running at boot or else you'll have no input. This will not work if you are using full-disk encryption.

I am referring to the keyboard. If only the headset appears in the output of the detect script, it is because the keyboard cannot be seen. Driver unloaded or keyboard locked up.

vonschutter, 2017-12-06:
Please see my follow up mail. I could only get the K70 to work intermittently. I foude documentation on the readme.md stating that the driver may not be possible to use if you have an encrypted HD: If it still doesn't work, try replacing `0x20000408` with `0x4`. Note that this will cause the kernel driver to ignore the device(s) completely, so you need to ensure ckb-daemon is running at boot or else you'll have no input. This will not work if you are using full-disk encryption.

Sent from Mailspring (https://link.getmailspring.com/link/local-a21eea1d-1de0@Seth/0?redirect=https%3A%2F%2Fgetmailspring.com%3Fref%3Dclient&recipient=reply%2B020ae6f999f761194b533b009bc378507e1f2cab4ad1d4dc92cf00000001163e47c992a169ce10a5f3bb%40reply.github.com), the best free email app for work

ZirconiumX, 2017-12-06:
>If only the headset appears in the output of the detect script, it is because the keyboard cannot be seen. Driver unloaded or keyboard locked up.

Even if there's no driver for a device, it should still be present on the USB bus.

Quirk 0x4 is for devices completely unsupported by the Linux `usbhid` driver. Your keyboard is very much supported, so you should use the 0x20000408 quirks. That will give you access to the encryption prompt through the kernel `usbhid` driver until ckb-daemon takes over at boot.

vonschutter, 2017-12-06:
OK thank you Dan. I must have done something wrong then... Will try again... Thank you for your prompt answers.

Sent from Mailspring (https://link.getmailspring.com/link/local-39e5e65d-adff@Seth/0?redirect=https%3A%2F%2Fgetmailspring.com%3Fref%3Dclient&recipient=reply%2B020ae6f9ea13acabce7e20a60622a5dc46d82ebf6839d1b592cf00000001163f6ac292a169ce10a5f3bb%40reply.github.com), the best free email app for work

closed 2018-01-21 by tatokis