mattanger/ckb-next #359: K63 swapped multimedia keys (led only)

aktentasche, 2017-12-05:
Hi,

so I am not sure if this an issue with the keyboards firmware (did not find an option to upgrade it in Windows application):

The multimedia keys on top work as expected, however, "volume up" and "volume down" never light up. Through animations I found out that "stop" and "skip back" light up in their place, confirmed by manually setting the light level in the ckb GUI. 

Now, I'd be happy to debug this in the source code and create a PR if a dev could point me to a source file where I can start :)

Currently I'm using 0.2.8 via https://aur.archlinux.org/packages/ckb-next-git

Thanks!

ZirconiumX, 2017-12-06:
First things first - can you reproduce this using the ckb-next-latest-git package? That gives you the very latest source code.

If you really want to dig in, [join us in IRC](http://webchat.freenode.net/?channels=%23ckb-next&uio=d4), since what you need to do is a little hard to explain on GitHub. The file you're looking for is the keymap array in src/ckb-daemon/keymap.c, but you'll need to poke the daemon directly to test your lights.

aktentasche, 2017-12-10:
I could reproduce it with your suggested AUR package.

Looking at the keymap.c I found that the LED code for volup and voldown were already in use. Interestingly, I did not find an earlier version where it is different. However, looking at how the keymap array is structured, I could come up with the proper LED codes for volup and voldown.

Tested and working on a K63 compact, using gdbserver+QT creator.

I will create a PR (my first one ever :+1:)

closed 2017-12-10 by aktentasche