mattanger/ckb-next #363: Cannot find file: .

kfirufk, 2017-12-15:
Hi! :) every week or so I would pull the latest source in master branch, re-execute quickinstall and i'm all done.

now when I try to execute quickinstall I get the following output:
```
Command line tools are already installed.
Preparing build files...
Empty filename passed to function
Cannot find file: .

Press enter to exit.
```

I tried executing `make distclean && ./qmake-auto && ./quickinstall` but same results.

try re-cloning an reinstalling but same results.

on a Mac Pro with HighSierra 10.13.2 with xcode 9.2 and an amazing Corsair Gaming K95 RGB PLATINUM Keyboard.

[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1564177/ckb-dev-detect-report.gz)

light2yellow, 2017-12-16:
> a Mac Pro with HighSierra 10.13.2 with xcode 9.2

I have the exact same setup (Hackintosh though, but it shouldn't matter) and cannot reproduce.

Make sure to:
* have XCode installed
* have XCode launched for at least once
* have Command Line Tools chosen in XCode (there was a recent update of CLT from App Store)

After that, run `bash -xv ./quickinstall` and post the output. This will run the script in debug mode.

kfirufk, 2017-12-16:
I tried to reinstall command line tools for xcode 9.2 but the results are the same.

output of quickinstall in debug mode:

```
#!/bin/bash
# ckb auto build/install script

# Temporary files for program output
TMPFILE=$(mktemp /tmp/ckb.XXXXXXXX)
mktemp /tmp/ckb.XXXXXXXX
++ mktemp /tmp/ckb.XXXXXXXX
+ TMPFILE=/tmp/ckb.SxKJChWO
function newtmp {
    rm -f "$TMPFILE"
    TMPFILE=$(mktemp /tmp/ckb.XXXXXXXX)
}

# Check for failure, quit if unsuccessful
function checkfail {
    if [[ $1 -ne 0 ]]; then
        cat "$TMPFILE"
        finish "$1"
    fi
}

# Exit function
function finish {
    rm -f "$TMPFILE"
    echo ""
    echo "Press enter to exit."
    read -rs dummy
    exit "$1"
}

# Directories
cd "$(dirname "$0")" || exit
dirname "$0"
++ dirname ./quickinstall
+ cd .
SRCDIR="src"
+ SRCDIR=src
BINDIR="bin"
+ BINDIR=bin
[[ "$OSTYPE" == "darwin"* ]] && BINDIR="ckb.app"
+ [[ darwin17 == \d\a\r\w\i\n* ]]
+ BINDIR=ckb.app
# This script respects the PREFIX variable on Linux (always /Applications on OSX)
[[ $PREFIX == "" ]] && PREFIX="/usr"
+ [[ '' == '' ]]
+ PREFIX=/usr
[[ "$OSTYPE" == "darwin"* ]] && PREFIX="/Applications"
+ [[ darwin17 == \d\a\r\w\i\n* ]]
+ PREFIX=/Applications

tab=$'\t'
+ tab=' '
bad_chars=" |'|$tab" # characters needed to be escaped
+ bad_chars=' |'\''|    '
if [[ "$(pwd)" =~ $bad_chars ]]; then
    echo "Error: you must compile ckb in a directory with no spaces/single quotes/tabs."
    echo "Move or rename the folder and try again."
    finish 1
fi
pwd
++ pwd
+ [[ /Volumes/osx-storage/src/ckb-next =~  |'|   ]]

# Install command line tools if missing
if [[ "$OSTYPE" == "darwin"* ]]; then
    if xcode-select --install 2>&1 | grep -q "installed"; then
        echo "Command line tools are already installed."
    else
        # check if user installed them after the gui prompt and prompt him once again
        if xcode-select --install 2>&1 | grep -q "installed"; then
            echo "Command line tools are installed."
        else
            echo "Please install command line tools and rerun the script."
            finish 1
        fi
    fi
    # Check for Xcode presence
    if xcodebuild -version 2>&1 | grep "error"; then
        echo "Please install Xcode and rerun the script."
        finish 1
    fi
fi
+ [[ darwin17 == \d\a\r\w\i\n* ]]
+ xcode-select --install
+ grep -q installed
+ echo 'Command line tools are already installed.'
Command line tools are already installed.
+ xcodebuild -version
+ grep error

# Purge old animations on Linux
if [ -d "/usr/bin/ckb-animations" ]; then
    echo "Found deprecated installation scheme, fixing"
    sudo rm -rf "/usr/bin/ckb-animations"
    checkfail $?
    echo "Done"
fi
+ '[' -d /usr/bin/ckb-animations ']'

if [[ -d $SRCDIR ]]; then
    echo "Preparing build files..."
    # Clean up first
    make clean >/dev/null 2>&1
    rm -f Makefile Makefile.* ./*/*/Makefile ./*/*/Makefile.* >/dev/null 2>&1
    # Generate makefiles
    ./qmake-auto "$QMAKEFLAGS" > "$TMPFILE" 2>&1
    checkfail $?
    make clean >/dev/null 2>&1
    # Run make with command arguments
    echo "Compiling binaries..."
    echo "(This can take a while, please be patient)"
    newtmp
    # Find out how many cores the system has, for make
    JOBS=$(getconf _NPROCESSORS_ONLN 2>/dev/null)
    # Default to 2 jobs if something went wrong earlier
    if [[ -z "$JOBS" ]]; then
        JOBS=2
    fi
    make -j "$JOBS" "$@" 2>"$TMPFILE"
    checkfail $?
    echo ""
    echo "Finished!"
fi
+ [[ -d src ]]
+ echo 'Preparing build files...'
Preparing build files...
+ make clean
+ rm -f Makefile 'Makefile.*' './*/*/Makefile' './*/*/Makefile.*'
+ ./qmake-auto ''
+ checkfail 2
+ [[ 2 -ne 0 ]]
+ cat /tmp/ckb.SxKJChWO
Empty filename passed to function
Cannot find file: .
+ finish 2
+ rm -f /tmp/ckb.SxKJChWO
+ echo ''

+ echo 'Press enter to exit.'
Press enter to exit.
+ read -rs dummy
```

ofirmiron, 2017-12-17:
I am getting the same error. Running High Sierra (10.13.2 - 17C88).
I installed Xcode and the Command Line Tools, checked for updates, as well as ensured that the Command Line Tools was selected in the Xcode settings.

commit 0cec46d9 (2017-12-17): quickinstall: fix #363
closed 2017-12-17 by light2yellow