mattanger/ckb-next #369: Corsair Glaive Mouse Delay.

ruixe, 2017-12-21:
Hi everyone, I am new to the GitHub thing... I am unsure about the format, please pardon me about this. I am a Linux (Lubuntu 17.04) user and I need help because my mouse, Glaive RGB sometimes shows delays when I scroll the wheel. It also sometimes gives delays when I click. This made gaming very uncomfortable. I am also new to codes and stuff. Please patiently tell me what to do because I am unsure of what is going on. Thank you!

ZirconiumX, 2017-12-21:
Sorry we missed you in IRC! We were all asleep (the dev team is mostly European).

Latency issues are very hard to fix, because ckb spends most of its time waiting for events from the kernel, which is expensive.

The lowest-latency approach is to write a kernel driver for Linux, but that's out of scope for this project.

At the moment, the driver sleeps on events from the kernel - this means the kernel has to give us a free timeslot, which has been known to be a latency problem on heavily loaded systems...such as somebody gaming.

A lower-latency approach is to constantly poll the device for packets, but this has the drawback of increased TL;DR.

Depending on how we choose to fix #367, this may or may not get fixed. But this is an unfortunate case of not being able to eat our cake and have it too.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/53

closed 2018-01-22 by tatokis