mattanger/ckb-next #370: Clicking on change color freezes ckb

peacockethan, 2017-12-25:
Using Ubuntu 17.10 with K70 RGB and M65 Pro, All demos work.  Installed with quick install.  When clicking on color value or change color to assign a new color in all locations ckb freezes.

Here is terminal ouput

```Console
ethanpeacock@xps-15:~$ ckb
QObject::startTimer: Timers cannot have negative intervals
Path  to settings is "/home/ethanpeacock/.config/ckb/ckb.conf"

(ckb:4670): GLib-GObject-WARNING **: cannot register existing type 'GtkWidget'

(ckb:4670): GLib-GObject-WARNING **: cannot add class private field to invalid type '<invalid>'

(ckb:4670): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(ckb:4670): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:4670): GLib-GObject-WARNING **: cannot register existing type 'GtkBuildable'

(ckb:4670): GLib-GObject-CRITICAL **: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed

(ckb:4670): GLib-CRITICAL **: g_once_init_leave: assertion 'result != 0' failed

(ckb:4670): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:4670): GLib-GObject-CRITICAL **: g_type_register_static: assertion 'parent_type > 0' failed

(ckb:4670): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'
^C

```

ZirconiumX, 2017-12-25:
https://github.com/mattanger/ckb-next#fedora-26-color-changer-freeze-fix

xdotax, 2017-12-27:
Thanks for this thread. If anyone else has this problem in 17.04, I had to install the Webupd8 PPA first, from here: https://www.ubuntuupdates.org/package/webupd8/zesty/main/base/qt5ct

```
sudo add-apt-repository ppa:nilarimogard/webupd8
sudo apt-get update
sudo apt install qt5ct`

light2yellow, 2017-12-27:
Duplicate of #70

closed 2017-12-27 by light2yellow