mattanger/ckb-next #372: Lose mouse cursor after quickinstall

paolo-g, 2017-12-28:
- [Devices information] Sabre RGB

- [Bug report]
    - [General information]
        - [Source] git clone
        - [Branch] master and I also tried v0.2.8 tag
    - [Logs & crash reports] quickinstall ran without err
    - [What is the program's current behavior?] Immediately after quickinstall starts the daemon, I lose my mouse. Restarting the computer doesn't help. Stopping the daemon gets me my mouse back
    - [Any other notes & comments?] I have Ubuntu 16 and
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1591384/ckb-dev-detect-report.gz)

 an encrypted hard drive

# Bug report

- [x ] Yes
- [ ] No

ZirconiumX, 2017-12-28:
This sounds like a communication failure between the daemon and the mouse. Hmm.

Can you [join us in IRC](http://webchat.freenode.net/?channels=%23ckb-next&uio=d4) please?

paolo-g, 2017-12-29:
Turns out the mouse was defective

closed 2017-12-29 by paolo-g