mattanger/ckb-next #373: CKB-v0.2.8.pkg Keypress hold with mouse click not working unless reinstall after each reboot

kaekae84, 2017-12-28:
__Switch on the "Preview" mode for better reading experience.__
Use it to put ticks and then switch back to write in the details where needed.

- [Before you proceed](#before-you-proceed)
- [Devices information](#devices-information)
- [New device support request](#new-device-support-request)
- [Feature request](#feature-request)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior)
    - [What would you like it to do instead?](#what-would-you-like-it-to-do-instead)
    - [Any other notes & comments?](#any-other-notes--comments)
- [Bug report](#bug-report)
    - [General information](#general-information)
        - [Source](#source)
        - [Branch](#branch)
    - [Logs & crash reports](#logs--crash-reports)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior-1)
    - [What is the expected behavior of the program?](#what-is-the-expected-behavior-of-the-program)
    - [Any other notes & comments?](#any-other-notes--comments-1)
- [Something else](#something-else)

---

# Before you proceed

- [ ] __Try__ the latest code by cloning and building the project manually.
- [ ] __Read__ the [`README.md`](https://github.com/mattanger/ckb-next/blob/master/README.md) _in full_.
- [ ] __Skim__ through other `.md` documents in the project's root. They contain more specific information.
- [ ] __Search__ in open _and_ closed [Issues](https://github.com/mattanger/ckb-next/issues). [Here's how](https://help.github.com/articles/searching-issues/).
- [ ] __Avoid__ opening a new issue in case a similar one already exists (don't worry, we get the notifications for closed issues as well, it will be reopened if needed).
- [ ] __Keep__ unrelated issues separate (create a new one if you are sure yours is unique).
- [ ] __Use__ [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to write on GitHub. Use triple backticks if copy-pasting more than _one_ line.

Now, __before you proceed even further__, make sure you are following the recommendations above. It will save a lot of time for you and us in the future.

---

# Devices information

Run [`ckb-dev-detect`](https://github.com/mattanger/ckb-next/blob/master/ckb-dev-detect) from the root of the source code tree:

```console
$ ./ckb-dev-detect
```

Upload `ckb-dev-detect-report.gz` on GitHub with this issue.

---

# New device support request

- [ ] Yes
- [ ] No

<sub><sup>leave empty if you have no new device support requests</sup></sub>

You should upload:
* __Linux__: `ckb-dev-detect-report.gz` is enough
* __macOS__: a full dump following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

- [ ] Yes
- [ ] No

<sub><sup>leave empty if you have no feature requests</sup></sub>

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

- [ ] Yes
- [ ] No

<sub><sup>leave empty if you have no bug reports</sup></sub>

## General information

### Source
(how and where did you get this program, e.g.: _"`ckb-next-git` package in AUR"_ or _"built manually using quickinstall script"_ or _"pkg for macOS"_)

### Branch
(fill in if ckb-next was compiled manually, e.g.: _"master"_ or _"newdev"_)

## Logs & crash reports
(any useful information - an OS crash report, debugger's backtrace, any non-standard way of usage (BIOS mode ...)  etc.)

You should upload:
- Linux: `ckb-dev-detect-report.gz` is enough
- macOS:
    * the output of `ckb-daemon`:
    ```console
    $ sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
    $ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ```
    * the output of `ckb` (open another terminal):
    ```console
    $ sudo killall -9 ckb
    $ /Applications/ckb.app/Contents/MacOS/ckb
    ```

## What is the program's current behavior?

Mac OS Sierra + CKB-V0.2.8 + Corsair K95 RGB Keyboard.

Unable to perform shift/cmd/alt + click after each reboot. 

However, the function came back once I reinstall the pkg file. Takes only a short while but I was wondering if there's a process that was killed by the system upon reboot and I could just restart that? I tried looking for the daemon file that most of us are talking about here but I couldn't find it anywhere in the Activity Monitor. I tried killing all the other process (all auto restarts) and it doesn't work. Only reinstalling the pkg works.

I work with images and those mouse click + key hold functions are crucial to fast work procedures...

Any help will be appreciated! 

Many thanks in advance!
KK

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

<sub><sup>if your issue is about something else, try your best to describe it here, otherwise leave empty</sup></sub>

Kukuruts, 2017-12-30:
I have the same issue on High Sierra with a Corsair Strafe RGB.
I can also confirm that reinstalling the pkg works for the current session.

kaekae84, 2018-01-01:
Does anyone know of a fix to this issue? Or is there anyone working on a solution? Is there a way I can learn to work on the code to get this straighten out too?

ZirconiumX, 2018-01-01:
The team is currently split between trying to add K55 support, adding CMake, and ironing out the protocol documentation. If you want to get this fixed yourself, join us in IRC, and we'll try our best to guide you.

ZirconiumX, 2018-01-01:
@tatokis has kindly informed me that the fix for this is already in Git master. You'll have to build ckb-next yourself, but that should fix it.

kaekae84, 2018-01-01:
Oh wow. Thank you @ZirconiumX ! So good to know I can do something about it. How do I actually build CKB next? Is there a easy to follow guide? I am completely new to this and honestly just know how to open that PKG file and install. Hopefully I can get some guidance!

Thank you in advance every one!

light2yellow, 2018-01-01:
@kaekae84 
> How do I actually build CKB next? Is there a easy to follow guide?

Follow the [README](https://github.com/mattanger/ckb-next#building-from-source).

kaekae84, 2018-01-01:
@light2yellow thank you. I checked the read me and that’s where I got the link to download the MAC os version as per mentioned in my first post. It is CKB Version 0.2.8 right? 

It resolves the issue at the beginning of every install/reinstall but once you reboot, it undo the keyboard + mouse commands functions availability. 

Am I missing some configuration steps here to make it permanent?

Thank you,
Kae

light2yellow, 2018-01-01:
@kaekae84 

as @ZirconiumX said, you'll have to compile the program. See [this](https://help.github.com/articles/cloning-a-repository/) to download the source code and follow the README - in the post above I linked to the section you are interested in.

MegaCookie, 2018-01-03:
I had the same issue but rebuilding fixed it! 
Who can update the binaries on Github?

kaekae84, 2018-01-05:
@ZirconiumX and @light2yellow thank you very much.:) with some fiddling around I managed it!

Does anyone know how I can update the binary? Perhaps I can make package of the new ckb next so people don’t have to compile it again?:)

kaekae84, 2018-01-05:
@MegaCookie thank you! A RL Friend of mine who is here in git guided me through the process!:D

closed 2018-01-21 by tatokis