mattanger/ckb-next #376: Command line interface

blshadow, 2018-01-08:
Hi. It is possible to control the keyboard from cli? It would be nice (for example, automate switching color profile, or for notifications), but I'm not find any documentation how to do it.

OvermindDL1, 2018-01-08:
As I recall you can send raw text commands to the proper `/dev/*` devices, so it is as simple as `echo "blah" > /dev/input/ckb#/cmd` or something like that, like on my computer:

```zsh
╰─➤  cd /dev/input
╰─➤  cat ckb0/connected 
/dev/input/ckb1 0702200EAE3E14A053BC1323F5001945 Corsair K95 RGB Gaming Keyboard
/dev/input/ckb2 04021031AEBA042055F2DF02F5001943 Corsair Gaming SCIMITAR RGB Mouse
╰─➤  cat ckb0/version  
v0.2.8
╰─➤  cat ckb1/features 
corsair k95 rgb pollrate bind notify fwversion fwupdate
╰─➤  cat ckb2/features 
corsair scimitar rgb pollrate adjrate bind notify fwversion fwupdate
```

And so forth.  It's all detailed at:  https://github.com/mattanger/ckb-next/wiki/CKB-Daemon-Manual

:-)

blshadow, 2018-01-08:
It's not possible to change color if ckb user application is active and mode with animation is selected.

ZirconiumX, 2018-01-08:
The daemon has no way of communicating to the GUI that the colour has been changed by somebody else, so you end up with a desynchronization between the two.

blshadow, 2018-01-08:
Then it would be nice to be able to control a ckb application from cli.

ZirconiumX, 2018-01-09:
I'll see what I can do, then. It's entirely possible it won't work alongside the GUI, but that's up to you.

OvermindDL1, 2018-01-09:
@ZirconiumX It would actually be nice if the GUI could 'suspend' it's operations when something else is using it, or if the something else would override the GUI settings until they reset it back to whatever the GUI is doing.

ZirconiumX, 2018-01-09:
@OvermindDL1 And how do you plan to detect that? We can't, at the moment.

fleischie, 2018-01-09:
I once started implementing a cli for ckb. It initially was supposed to start and controlthe GUI in the background.

It was scrapped at some point (i think) but I can't remember exactly why. It was a convincing reason enough to do so, though.

OvermindDL1, 2018-01-09:
@ZirconiumX There is a notification built into the driver, have it send a notification when something sends a command that uses, say, a unique key or PID.  If a key you could 'key' commands with a prefix to a command, so something like `<key> <cmd>...` so the ckb gui itself could be keyed like `ckb <cmd>...`, anytime something uses a new key then a notification is sent on a notify pipe.

_/me is bikeshedding about the above, is not happy about the API, would prefer a connection to be established via holding a pipe open_

ZirconiumX, 2018-01-09:
If we're going for that, we might as well just make it a dbus daemon and be done with it.

ZirconiumX, 2018-01-10:
...Judging by the reactions on that, you thought I was being serious.

Guess we should add that to the long, long list of issues I have to do. ;~;

OvermindDL1, 2018-01-10:
> ...Judging by the reactions on that, you thought I was being serious.

Lol, not a bad idea though, and making DBus daemons is not a hard thing by any stretch (though still non-trivial).  ^.^;

ZirconiumX, 2018-01-10:
@tatokis disagrees.

OvermindDL1, 2018-01-10:
> The plan might be to have the GUI dbus-controllable.

As long as it all allows temporary modifications, like a program causing a single red pulse to go out over the keyboard when a notification comes in.  :-)

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/54

closed 2018-01-22 by tatokis