mattanger/ckb-next #387: Sabre + K65 issues

wannesvc, 2018-09-14:
Hi,

when I plug in my K65 keyboard an sabre mouse at the same time, The ckb-next daemon crashes and my peripherals stop responding. I get the following error message:

```
-- The start-up result is RESULT.
Sep 14 09:10:25 vancawa.cloudar ckb-next-daemon[7808]:     ckb-next: Corsair RGB driver 0.3.1-48-ge8d644f
Sep 14 09:10:25 vancawa.cloudar ckb-next-daemon[7808]: [I] Root controller ready at /dev/input/ckb0
Sep 14 09:10:25 vancawa.cloudar ckb-next-daemon[7808]: [I] Connecting Corsair Gaming SABRE RGB Mouse at /dev/input/ckb1
Sep 14 09:10:25 vancawa.cloudar ckb-next-daemon[7808]: [I] Connecting Corsair K65 RGB Gaming Keyboard at /dev/input/ckb2
Sep 14 09:10:25 vancawa.cloudar kernel: input: ckb2: Corsair K65 RGB Gaming Keyboard vKB as /devices/virtual/input/input75
Sep 14 09:10:25 vancawa.cloudar kernel: input: ckb1: Corsair Gaming SABRE RGB Mouse vKB as /devices/virtual/input/input76
Sep 14 09:10:25 vancawa.cloudar kernel: input: ckb2: Corsair K65 RGB Gaming Keyboard vM as /devices/virtual/input/input77
Sep 14 09:10:25 vancawa.cloudar kernel: input: ckb1: Corsair Gaming SABRE RGB Mouse vM as /devices/virtual/input/input78
Sep 14 09:10:25 vancawa.cloudar ckb-next-daemon[7808]: [I] Starting input thread for /dev/input/ckb2
Sep 14 09:10:25 vancawa.cloudar ckb-next-daemon[7808]: [I] Starting input thread for /dev/input/ckb1
Sep 14 09:10:25 vancawa.cloudar systemd-logind[1194]: Watching system buttons on /dev/input/event18 (ckb1: Corsair Gaming SABRE RGB Mouse vKB)
Sep 14 09:10:25 vancawa.cloudar systemd-logind[1194]: Watching system buttons on /dev/input/event20 (ckb2: Corsair K65 RGB Gaming Keyboard vM)
Sep 14 09:10:25 vancawa.cloudar systemd-logind[1194]: Watching system buttons on /dev/input/event19 (ckb1: Corsair Gaming SABRE RGB Mouse vM)
Sep 14 09:10:25 vancawa.cloudar systemd-logind[1194]: Watching system buttons on /dev/input/event17 (ckb2: Corsair K65 RGB Gaming Keyboard vKB)
Sep 14 09:10:26 vancawa.cloudar ckb-next-daemon[7808]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 04 02 01 01 1c 1b 2f 1b 01 01 01 30 c1 ff 40 03 5e 00 01 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 01 01 00 01 00 00 6c >
Sep 14 09:10:26 vancawa.cloudar ckb-next-daemon[7808]: [I] Setup finished for /dev/input/ckb1
Sep 14 09:10:26 vancawa.cloudar audit[7808]: ANOM_ABEND auid=4294967295 uid=0 gid=0 ses=4294967295 subj=system_u:system_r:unconfined_service_t:s0 pid=7808 comm="ckb-next-daemon" exe="/usr/local/bin/ckb-next-daemon" sig=11 res=1
Sep 14 09:10:26 vancawa.cloudar ckb-next-daemon[7808]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 05 02 11 00 1c 1b 17 1b 01 01 01 06 c0 ff 40 00 03 00 00 00 02 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 >
Sep 14 09:10:26 vancawa.cloudar kernel: ckb-next-daemon[7811]: segfault at 0 ip 00000000004038ba sp 00007f84e7ffed40 error 4 in ckb-next-daemon[400000+18000]
Sep 14 09:10:26 vancawa.cloudar systemd[1]: ckb-next-daemon.service: Main process exited, code=killed, status=11/SEGV
Sep 14 09:10:26 vancawa.cloudar systemd[1]: ckb-next-daemon.service: Failed with result 'signal'.
Sep 14 09:10:26 vancawa.cloudar audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=ckb-next-daemon comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=fa>
Sep 14 09:10:26 vancawa.cloudar systemd[1]: ckb-next-daemon.service: Service hold-off time over, scheduling restart.
Sep 14 09:10:26 vancawa.cloudar systemd[1]: ckb-next-daemon.service: Scheduled restart job, restart counter is at 5.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Automatic restarting of the unit ckb-next-daemon.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Sep 14 09:10:26 vancawa.cloudar systemd[1]: Stopped Corsair Keyboards and Mice Daemon.
-- Subject: Unit ckb-next-daemon.service has finished shutting down
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit ckb-next-daemon.service has finished shutting down.
Sep 14 09:10:26 vancawa.cloudar systemd[1]: ckb-next-daemon.service: Start request repeated too quickly.
Sep 14 09:10:26 vancawa.cloudar systemd[1]: ckb-next-daemon.service: Failed with result 'signal'.
Sep 14 09:10:26 vancawa.cloudar systemd[1]: Failed to start Corsair Keyboards and Mice Daemon.
-- Subject: Unit ckb-next-daemon.service has failed
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit ckb-next-daemon.service has failed.
```

When I use each device separately I don't experience any issues. Could you please help me out? (thanks for the work on ckb-next!!)

ZirconiumX, 2018-09-14:
You really should have posted this in ckb-next/ckb-next like the entire repo tells you to do.

wannesvc, 2018-09-14:
sorry about that, when I google ckb-next, this repo is the first to come up, I'll post it over there

closed 2018-09-14 by wannesvc