mattanger/ckb-next #388: Harpoon Mice

kim2325, 2019-03-22:
Corsair Harpoon RGB Mice stopped working after installing ckb. I uninstalled the package, re-plugged to a different USB port, restarted my Mac, and it still does not respond. The light is blinking which identifies that it is plugged in but no longer works.

tatokis, 2019-03-22:
> THIS REPOSITORY HAS MOVED

Install the latest version from https://github.com/ckb-next/ckb-next/
https://github.com/ckb-next/ckb-next/releases/download/v0.4.0/ckb-next_v0.4.0.dmg

closed 2019-03-22 by tatokis

kim2325, 2019-03-22:
I did install the latest version and it made my mice stop working

tatokis, 2019-03-22:
Please open a new issue on the new tracker and follow the instructions to generate and attach a report.
https://github.com/ckb-next/ckb-next/issues/new